TerosTechnology/vscode-terosHDL #2: References to similar projects

umarcor, 2020-06-05:
[dbhi/vsc-hdl](https://github.com/dbhi/vsc-hdl) is a placeholder/playground where I've been gathering references/ideas and trying features provided by VSCode Extensions to achieve a solution similar to the purpose of this repo. I think it would be desirable to add a brief explanation to the README referencing [VUnit Test Explorer](https://marketplace.visualstudio.com/items?itemName=hbohlin.vunit-test-explorer) and [hackfin.gitlab.io/xhdl](https://hackfin.gitlab.io/xhdl/), since both are projects that could be potentially integrated here.

smgl9, 2020-06-06:
Interesting. 
I think that the features of those projects are already in development under TerosTechnology/colibri
This repository is brand new but yes, it should have a decent readme for the release, explaining its ultimate aim.
VUnit Test Explorer looks like TerosHdl v1 !

umarcor, 2020-06-06:
> I think that the features of those projects are already in development under TerosTechnology/colibri

Precisely, my main purpose is to share concepts/ideas, because I see that there is already some effort duplication. We will then see if codebases can be actually integrated.

umarcor, 2020-06-17:
FTR, the May 2020 update of VSCode allows "grouping views": 

> You might also want to group some views together that come from different extensions or you feel the default groups of built-in views aren't quite right for you. You can both move views into existing groups or create new groups for a select set of views. This works across the Side Bar and Panel just as before.

qarlosalberto, 2020-06-17:
Nice! Although it seems a feature for the user, not for the API. 

Have you seen the TerosHDL extension features? https://github.com/TerosTechnology/vscode-terosHDL/blob/master/README.md 
We will launch a release at the end of the month.

umarcor, 2020-06-25:
> Nice! Although it seems a feature for the user, not for the API.

It's a feature for the user, but the Extension might need to support it. See "For extension authors contributing views or view containers" in https://code.visualstudio.com/updates/v1_46#_flexible-layout
 
> Have you seen the TerosHDL extension features? [/README.md@`master`](https://github.com/TerosTechnology/vscode-terosHDL/blob/master/README.md?rgh-link-date=2020-06-17T18%3A55%3A20Z)
> We will launch a release at the end of the month.

I'm looking forward to trying it as soon as it is released ;)

closed 2023-03-15 by qarlosalberto