TerosTechnology/vscode-terosHDL #3: The following functions do not work in version 0.0.1.

AriesKo, 2020-06-22:
Warning occurs for each function as shown below.
Go to definition
-  [image]

Module documentation
- Nothing displayed

Generate template
- Select a valid file.!

Dependencies viewer
- Please, install Python 3.

qarlosalberto, 2020-06-22:
- What is your OS?
- Have you Python in the system path?
- What is the output if you execute: "python --version" and if you execute "python3 --version"?

AriesKo, 2020-06-22:
I'm using the windows10.
Python path is setting the in the environemnt variables box.
(PATH = C:\Program Files\Python38\Scripts\;C:\Program Files\Python38\;)
I updated python version today. (3.7.4 --> 3.8.3)
[image]

smgl9, 2020-06-22:
I've been able to reproduce the errors in Windows. In Linux the extension seems to be working fine. I have python 3.7 installed and don't have the "Please, install Python 3." error but... yes, something that was working before is broken for windows now. We'll try to figure it out. Thanks a lot for the report.

AriesKo, 2020-06-23:
Um... Then I'll run it on the Linux OS.

qarlosalberto, 2020-06-25:
@AriesKo we have fixed the errors in a new version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v0.0.2

AriesKo, 2020-06-26:
Thank you for your hard work
I have a quick feedback on the updated teroshdl.
I confirmed that the Module documentation, Generate template, and Dependencies viewer are working.
But Go to definition still doesn't work.

And could you report the result like vsg after using the format function?

qarlosalberto, 2020-06-27:
Do you say to update style checker errors "on save" and when you use the format function? If you want to use VSG you have to install the experimental branch (https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/401) -> https://github.com/jeremiah-c-leary/vhdl-style-guide/tree/issue-401

It does not work anything to go definition? For VHDL it's supported: ports, process, constants, signals and functions. Can you see the code structure (signals, ports...) in outline?

AriesKo, 2020-06-29:
VSG plans to utilize task.json until it is integrated into teroshdl.
The code that does not function as Go definition is as follows.
[image]

qarlosalberto, 2020-09-11:
Could you test it in the new beta release, 0.0.4?

AriesKo, 2020-09-14:
I have confirmed that the Go to definition function works. Thank you for your hard work.

closed 2020-09-14 by qarlosalberto