TerosTechnology/vscode-terosHDL #4: Packages not found in v0.0.1

lukipedio, 2020-06-23:
When I open a source file with use clause, I got the following error notification

[image]

qarlosalberto, 2020-06-23:
It's normal. You are using the GHDL linter.  The command -s (http://ghdl.free.fr/ghdl/Check-syntax-command.html) does both syntax and semantic checking. You can change the linter or disable it.

lukipedio, 2020-06-23:
Ok. I see.
I have menaged to change linter by manually editing the <userdir>\.vscode\extensions\teros-technology.teroshdl-0.0.1\package.json in Win10.
Is there a better way to do it?
I have changed from ghdl (Ihave not installed) to modelsim. Now I don't get the error anymore.

lukipedio, 2020-06-24:
Ok. At first I was not able to find the extension in the Extensions search. I missed to search in the @installed group.
So now I can modify the extension settings through the GUI instead of modifying the json file.

closed 2020-06-24 by lukipedio