TerosTechnology/vscode-terosHDL #5: Generate template error in v0.0.1

lukipedio, 2020-06-23:
If I click on the "Generate template" icon, I get the followeing error in the "Develop Tools console:
[image]

smgl9, 2020-06-23:
In Windows the extension is currently broken. And the error you mention is probably a change in an interface. We'll check it out but the extension is likely to be unstable until the official release. Especially in Windows but thanks a lot for the report. We'll let you know as soon as the errors are fixed.

lukipedio, 2020-06-24:
I will stop testing Windows extension until there will be a notification it is not broken anymore.
I will switch to Linux for further testing.

qarlosalberto, 2020-06-25:
@AriesKo we have fixed the errors in a new version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v0.0.2

closed 2020-06-27 by qarlosalberto