TerosTechnology/vscode-terosHDL #6: No documentation in v0.0.2

lukipedio, 2020-06-26:
I have installed v0.0.2 on a Linux Ubuntu 18.04.03 machine.

When I try to get the documentation of a VHDL module. I don't get it (no symbol, co comments).

Here is the developer tool log:

[Extension Host] Error: Command failed: python /home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/node_modules/jsteros/src/parser/parser.py "!" "/tmp/f-2020526-16125-1lt54i9.8xxak"	at checkExecSyncError (child_process.js:634:11)	at execSync (child_process.js:670:15)	at electron/js2c/asar.js:696:23	at VhdlParser.getAll (/home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/node_modules/jsteros/src/parser/vunitVhdlParser.js:43:20)	at Documenter._get_code_tree (/home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/node_modules/jsteros/src/documenter/documenter.js:265:34)	at Documenter._get_html_from_code (/home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/node_modules/jsteros/src/documenter/documenter.js:150:32)	at Documenter.get_html (/home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/node_modules/jsteros/src/documenter/documenter.js:70:31)	at Object.<anonymous> (/home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/out/lib/documenter/documenter.js:64:53)	at Generator.next (<anonymous>)	at fulfilled (/home/lcolombini/.vscode/extensions/teros-technology.teroshdl-0.0.2/out/lib/documenter/documenter.js:5:58)

Code : 1.46.0
Python2 : 2.7.17
Python3 :  3.6.9

lukipedio, 2020-06-26:
Correction: this issue is only on a single file (top level). Other VHDL files are correctly translated by Documenter.
The problematic file is syntactilly ok.

Here it is  (library declarations and entity):
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.user_dpp_pha_regs_pkg.all;
use work.V2740_PHA_PACK.all;

entity user_dpp is
  generic (
    -- Users to add parameters here
    C_NUM_CHANNELS : integer := 64;
    
    -- User parameters ends
    -- Do not modify the parameters beyond this line
    
    C_AXI_ADDR_WIDTH : integer                 := 32;  -- width of the AXI address bus
    C_BASEADDR : std_logic_vector(31 downto 0) := x"88000000"; -- the register file's system base address
    
    -- Parameters of Axi Slave Bus Interface S_DPP_REGS_AXI
    C_S_DPP_REGS_AXI_DATA_WIDTH : integer   := 32;
    C_S_DPP_REGS_AXI_ADDR_WIDTH : integer   := 32;
    
    -- Parameters of Axi Slave Bus Interface S_IN_DATA_AXIS
    C_S_IN_DATA_AXIS_TDATA_WIDTH    : integer   := 16;
    
    -- Parameters of Axi Master Bus Interface S_OUT_DATA_AXIS
    C_S_OUT_DATA_AXIS_TDATA_WIDTH   : integer   := 64
  );
  port (
    -- Users to add ports here
    
    -- User ports ends
    -- Do not modify the ports beyond this line
    
    
    -- Ports of Axi Slave Bus Interface S_DPP_REGS_AXI
    axi_aclk                : in std_logic;
    axi_aresetn             : in std_logic;
    s_dpp_regs_axi_awaddr   : in std_logic_vector(C_S_DPP_REGS_AXI_ADDR_WIDTH-1 downto 0);
    s_dpp_regs_axi_awprot   : in std_logic_vector(2 downto 0);
    s_dpp_regs_axi_awvalid  : in std_logic;
    s_dpp_regs_axi_awready  : out std_logic;
    s_dpp_regs_axi_wdata    : in std_logic_vector(C_S_DPP_REGS_AXI_DATA_WIDTH-1 downto 0);
    s_dpp_regs_axi_wstrb    : in std_logic_vector((C_S_DPP_REGS_AXI_DATA_WIDTH/8)-1 downto 0);
    s_dpp_regs_axi_wvalid   : in std_logic;
    s_dpp_regs_axi_wready   : out std_logic;
    s_dpp_regs_axi_bresp    : out std_logic_vector(1 downto 0);
    s_dpp_regs_axi_bvalid   : out std_logic;
    s_dpp_regs_axi_bready   : in std_logic;
    s_dpp_regs_axi_araddr   : in std_logic_vector(C_S_DPP_REGS_AXI_ADDR_WIDTH-1 downto 0);
    s_dpp_regs_axi_arprot   : in std_logic_vector(2 downto 0);
    s_dpp_regs_axi_arvalid  : in std_logic;
    s_dpp_regs_axi_arready  : out std_logic;
    s_dpp_regs_axi_rdata    : out std_logic_vector(C_S_DPP_REGS_AXI_DATA_WIDTH-1 downto 0);
    s_dpp_regs_axi_rresp    : out std_logic_vector(1 downto 0);
    s_dpp_regs_axi_rvalid   : out std_logic;
    s_dpp_regs_axi_rready   : in std_logic;
    
    -- Ports of Axi Slave Bus Interface S_IN_DATA_AXIS
    user_clk     : in std_logic;
    user_reset  : in std_logic;
    
    user_run      : in std_logic; -- TODO name????
    user_exttrg   : in std_logic; -- TODO name????
    test_en       : in std_logic; -- TODO name????
     
    s_in_data_axis_tdata    : in std_logic_vector(C_S_IN_DATA_AXIS_TDATA_WIDTH*C_NUM_CHANNELS-1 downto 0);
    s_in_data_axis_tvalid   : in std_logic;
    
    -- Ports of Axi Master Bus Interface M_OUT_DATA_AXIS
            
    m_out_dpp_axis_dt_tdata   : out std_logic_vector(C_S_OUT_DATA_AXIS_TDATA_WIDTH*C_NUM_CHANNELS-1 downto 0);
    m_out_dpp_axis_dt_tvalid  : out std_logic_vector(C_NUM_CHANNELS-1 downto 0);
    m_out_dpp_axis_dt_tlast   : out std_logic_vector(C_NUM_CHANNELS-1 downto 0);
    m_out_dpp_axis_dt_tready  : in std_logic_vector(C_NUM_CHANNELS-1 downto 0);
    
    m_out_dpp_axis_iw_tdata   : out std_logic_vector(C_S_OUT_DATA_AXIS_TDATA_WIDTH*C_NUM_CHANNELS-1 downto 0);
    m_out_dpp_axis_iw_tvalid  : out std_logic_vector(C_NUM_CHANNELS-1 downto 0);
    m_out_dpp_axis_iw_tlast   : out std_logic_vector(C_NUM_CHANNELS-1 downto 0);
    m_out_dpp_axis_iw_tready  : in std_logic_vector(C_NUM_CHANNELS-1 downto 0);
    
    mem_dpp_freeloc        : in std_logic_vector(C_NUM_CHANNELS*13-1 downto 0)
        
 --       -- Trigger interface
 --       user_trg_o              : out std_logic
    );
end user_dpp;

qarlosalberto, 2020-06-27:
Thank you! I will check it

lukipedio, 2020-09-06:
fixed in v0.0.3

closed 2020-09-06 by lukipedio