TerosTechnology/vscode-terosHDL #7: Module documentation error in 0.0.2

AriesKo, 2020-06-26:
There are the following problems when working with documentation.
If there are less than 8 variables declared in Port
[image]
When there are 8 variables declared in Port
[image]
And when running Markdown, PDF, Image, HTML Export, the following error message occurs
[image]

qarlosalberto, 2020-06-27:
@smgl9 it sounds as this issue https://github.com/TerosTechnology/colibri/issues/69
Could you take a look?

smgl9, 2020-06-27:
Yes, they are probably related. I'm on it

smgl9, 2020-06-27:
@AriesKo  It seems that the issue comes from the fact that the module has no outputs and not the number of the ports. 
It is allowed not having inputs but without outputs the error came up. 
We'll fix it. And if there is another error trigger let us now. Thank you!

AriesKo, 2020-06-29:
Thank for your help.
I had the following problem while using the documentation function.
If the number of declared ports is small, the square box is not created properly.
[image]

smgl9, 2020-06-29:
Ok, this is related to https://github.com/TerosTechnology/colibri/issues/69

dra5546, 2020-08-15:
I'm also running into the
"Extention host terminated unexpectantly"
When Trying to export documentation in any of the formats (Markdown, PDF, HTML, and Image).

Note: This is Windows 10

qarlosalberto, 2020-08-15:
We are working in the issue :)

smgl9, 2020-08-21:
The diagram has been corrected and will be working fine in the next release.

AriesKo, 2020-08-27:
I have check that it works without problems. Adding a wavedrom is fantastic.
Do you have any plans to add a state machine?

smgl9, 2020-08-27:
Thank you @AriesKo 
In fact, we had an experimental implementation of state machines viewer in the Atom plug in, but it's a complex issue. It needs more  development.