TerosTechnology/vscode-terosHDL #9: Format not well set for case operator

lukipedio, 2020-07-24:
See example below
[image]

qarlosalberto, 2020-10-18:
@lukipedio could you share the code (no image)? So I can test it.

lukipedio, 2020-10-26:
Sure! Sorry I missed the notification.
[src.zip](https://github.com/TerosTechnology/vscode-terosHDL/files/5437584/src.zip)

What I complain about is:
- The state machine format: some case have a different indentation/formatting
- port and signal name are uppercase (itis not my default choice): I think this could be configured by vsg

Thanks for checking.

Using Tesroshdl 0.4

qarlosalberto, 2020-11-02:
In the new version (v0.0.5) port and signal name aren't uppercase