TerosTechnology/vscode-terosHDL #12: parser problem with wavedrom

evgeniyBolnov, 2020-08-24:
if you add a wavedrom description before the module description, then the parser breaks

```verilog
//! { signal: [
//!   ['sink',
//!     {  name: 'ready',       wave: '1........'},
//!     {  name: 'valid',       wave: '01......0'},
//!     {  name: 'sop',         wave: '010......'},
//!    	{  name: 'eop',         wave: '0......10'},
//!    	{  name: 'empty[3:0]', 	wave: 'x3......x', data: 'empty[3:0]'},
//!    	{  name: 'data[95:0]',  wave: 'x4......x', data: 'data[95:0]'},
//!   ],
//!   {},
//!   ['source',
//!     {  name: 'ready',       wave: '1........'},
//!     {  name: 'valid',       wave: '01......0'},
//!     {  name: 'sop',         wave: '010......'},
//!    	{  name: 'eop',         wave: '0......10'},
//!    	{  name: 'empty[3:0]', 	wave: 'x3......x', data: 'empty[3:0]'},
//!    	{  name: 'data[127:0]',  wave: 'x4......x', data: 'data[127:0]'},
//!   ]
//! ]}

module conv_6to8 (
  
  input         reset_n,  //! Don't used
  input         clk,      //! Don't used
```
[image]

Without wavedrom

[image]

smgl9, 2020-08-24:
Thanks for your feedback! This has been tested mainly in vhdl. We will fix it.

evgeniyBolnov, 2020-08-28:
I solved this problem, but I think this solution is temporary

Delete all spaces between comment and module declaration
```verilog
//!    	{  name: 'data[127:0]',  wave: 'x4......x', data: 'data[127:0]'},
//!   ]
//! ]}
module conv_6to8 (
  
  input         reset_n,  //! Don't used
  input         clk,      //! Don't used
```

smgl9, 2020-08-29:
That's right. In the test I did I got the same result. But we should  be able to put spaces wherever we desire. This was working before.

qarlosalberto, 2020-09-11:
Could you test it in the new beta release, 0.0.4?

evgeniyBolnov, 2020-09-12:
> Could you test it in the new beta release, 0.0.4?

Looks great, thanks. I tried to break it - it didn't work

closed 2020-09-12 by qarlosalberto