TerosTechnology/vscode-terosHDL #13: Add SystemVerilog support

evgeniyBolnov, 2020-08-25:
The plugin only works with files with the v extension, can you add sv?
[image]
[image]

smgl9, 2020-08-25:
In general there are errors with Verilog/SystemVerilog. We are working to have them solved ASAP.
Thanks for the feedback. It's very useful.

qarlosalberto, 2020-11-02:
Could you check the release (v0.0.5)? We have added SV support.

evgeniyBolnov, 2020-11-03:
> Could you check the new release (v0.0.5)? We have added SV support.

[image]

qarlosalberto, 2020-11-03:
I uploaded a wrong version, download it again.

closed 2020-12-10 by qarlosalberto