TerosTechnology/vscode-terosHDL #15: Error - Select a valid file.!

douglasrossi, 2020-08-28:
Whenever I click on `Generate template`, I get an error notification:

> Select a valid file.!

Also, I can't see any diagrams in `Module documentation preview`.

Ubuntu 20.04
Visual Studio Code 1.48.2 (Snap)
Python 3.8.2

qarlosalberto, 2020-08-28:
Is It the 0.0.3 version or 0.0.2? 
What is your file extension?

douglasrossi, 2020-08-28:
Version 0.0.3, using only `.vhd` files.
Unlike other functions, the `Open dependencies viewer` is working when I add my project files.

qarlosalberto, 2020-08-28:
Do you see any error in the debug console?

douglasrossi, 2020-08-28:
I see no message in the `Debug console`. It seems that even the `Format` function is not working.

jiv4ik, 2020-08-28:
I have the same bug.  My debug console log:

notificationsAlerts.ts:40 Select a valid file.!
onDidChangeNotification @ notificationsAlerts.ts:40
(anonymous) @ notificationsAlerts.ts:26
fire @ event.ts:587
addNotification @ notifications.ts:207
notify @ notificationService.ts:106
(anonymous) @ mainThreadMessageService.ts:83
_showMessage @ mainThreadMessageService.ts:44
$showMessage @ mainThreadMessageService.ts:38
_doInvokeHandler @ rpcProtocol.ts:409
_invokeHandler @ rpcProtocol.ts:394
_receiveRequest @ rpcProtocol.ts:310
_receiveOneMessage @ rpcProtocol.ts:237
(anonymous) @ rpcProtocol.ts:105
fire @ event.ts:587
fire @ ipc.net.ts:463
_receiveMessage @ ipc.net.ts:747
(anonymous) @ ipc.net.ts:602
fire @ event.ts:587
acceptChunk @ ipc.net.ts:240
(anonymous) @ ipc.net.ts:201
t @ ipc.net.ts:29
emit @ events.js:203
addChunk @ _stream_readable.js:295
readableAddChunk @ _stream_readable.js:276
Readable.push @ _stream_readable.js:210
onStreamRead @ internal/stream_base_commons.js:166

qarlosalberto, 2020-08-28:
@jiv4ik what is your Python version and OS?

jiv4ik, 2020-08-28:
> @jiv4ik what is your Python version and OS?
Ubuntu 18.04.5 LTS
Python 3.6.9
Python 2.7.17
Visual Studio Code Version: 1.48.2

smgl9, 2020-08-28:
Mmm... I have an identical configuration, but it works for me. Could you share a file that is not working for you?

jiv4ik, 2020-09-01:
I have solved this problem for myself. I removed comments after the end of the module. You can't write like that:
module test(
           input clk, //! clk           
           input reset_n,
           output  out_a
       );
endmodule

// comment

smgl9, 2020-09-01:
ok! 

We are testing a solution for these errors that will be available in the next beta release. And it seems to solve all the mentioned problems.

qarlosalberto, 2020-09-01:
@douglasrossi can you share a file (or a reduced file) that fails?

douglasrossi, 2020-09-01:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_misc.all;

entity cs is
  port (
    -- sel
    sel_i : in  std_logic_vector(3 downto 0);
    -- switch
    cmd_i : in  std_logic_vector(3 downto 0);
    cmd_o : out std_logic
    );
end cs;

architecture rtl of cs is

begin

  cmd_o <= or_reduce(sel_i and cmd_i);

end rtl;
```

qarlosalberto, 2020-09-11:
Could you test it in the new beta release, 0.0.4?

douglasrossi, 2020-09-11:
> Could you test it in the new beta release, 0.0.4?

Same problem.
`Select a valid file.!`

jools76, 2020-09-21:
I have same problem too.

jools76, 2020-09-21:
**python**
python3 version is 3.7.1 
Windows7 x64. Python from conda.

qarlosalberto, 2020-09-21:
The dependencies viewer work fine for you?

jools76, 2020-09-21:
No. It don't work.

qarlosalberto, 2020-09-21:
@douglasrossi does dependencies viewer work for you?

qarlosalberto, 2020-09-21:
@jools76 if you open a terminal what is the output of **python --version**

jools76, 2020-09-21:
Python 3.7.1

jools76, 2020-09-21:
It's all good! 
Python interpreter must be in the PATH.

douglasrossi, 2020-09-21:
> @douglasrossi does dependencies viewer work for you?

Yes, it works.

qarlosalberto, 2020-11-02:
@douglasrossi could you check the new version (v0.0.5). I hope it works

douglasrossi, 2020-11-03:
@qarlosalberto , now it's working.
I'm kind of missing a shortcut to the dependency viewer, but that's another issue.
Thank you!

closed 2020-11-03 by douglasrossi