TerosTechnology/vscode-terosHDL #16: Module documentation error

jiv4ik, 2020-08-28:
I try to generate module documentation, but module documentation window remains empty.
i see in debug console:
console.ts:137 [Extension Host] TypeError: Cannot read property 'substr' of undefined	at tsVerilogParser.extractData (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/parser/tsVerilogParser.js:338:42)	at tsVerilogParser.getComments (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/parser/tsVerilogParser.js:222:48)	at tsVerilogParser.getGenerics (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/parser/tsVerilogParser.js:243:25)	at tsVerilogParser.getAll (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/parser/tsVerilogParser.js:54:26)	at async Documenter._get_code_tree (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/documenter/documenter.js:290:21)	at async Documenter._get_html_from_code (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/documenter/documenter.js:153:21)	at async Documenter.get_html (/home/anton/.vscode/extensions/teros-technology.teroshdl-0.0.3/node_modules/jsteros/src/documenter/documenter.js:71:20)

smgl9, 2020-08-28:
Thank you for your feedback! 
I guess that is a verilog file. I think many of the verilog issues are related. #12

jiv4ik, 2020-09-01:
when i removed comments after endmodule (https://github.com/TerosTechnology/vscode-terosHDL/issues/15#issuecomment-684762956) i saw the module documentation. 
And i have the same problem with blank lines

closed 2020-09-01 by jiv4ik