TerosTechnology/vscode-terosHDL #17: Feature Request: BitField Support

dra5546, 2020-08-28:
Given that TerosHDL already has Wavedrom support (which is awesome by the way!), there is a projects authored by the same person that provide some neat diagrams in much the same way as Wavedrom is constructed. Would it be possible to support [BitField](https://github.com/wavedrom/bitfield) in much the same way Wavedrom is supported already?

A way to see a BitField diagram in action would be to put to put the following text in the form at https://wavedrom.com/editor.html

{reg: [
    {bits: 7,  name: 'opcode',    attr: 'OP-IMM'},
    {bits: 5,  name: 'rd',        attr: 'dest'},
    {bits: 3,  name: 'func3',     attr: ['ADDI', 'SLTI', 'SLTIU', 'ANDI', 'ORI', 'XORI'], type: 4},
    {bits: 5,  name: 'rs1',       attr: 'src'},
    {bits: 12, name: 'imm[11:0]', attr: 'I-immediate[11:0]', type: 3}
]}

qarlosalberto, 2020-08-28:
It's a really good idea. I hope to add it in the next beta version. Maybe the next week

qarlosalberto, 2020-09-10:
@dra5546 It's okay like that?

[image]

dra5546, 2020-09-10:
Perfect! Thanks for adding BitField!

qarlosalberto, 2020-09-11:
Could you test it in the new beta release, 0.0.4?

dra5546, 2020-09-12:
@qarlosalberto Works perfectly as far as I can tell. Couldn't break it.

closed 2020-09-12 by qarlosalberto