TerosTechnology/vscode-terosHDL #46: Port documentation

GlenNicholls, 2020-11-16:
For the documentation of VHDL/V/SV, it would be nice if you can choose where this goes. Take the following example:

```
--! This is a description
--! of the entity.
entity counter is
  port (
    clk: in std_logic; --! Clock comment
    out_data: out std_logic --! Description **port comment**
  );
end counter;
```

I don't like documenting ports/generics in-line because it is a headache to deal with when instantiating an entity by copying/pasting the declaration. The other problem with commening generics/ports/subprogram parameters in-line is for anything complicated, it makes it really difficult to see what all is there because descriptions are more verbose. For complicated functionality in port/subprogram parameter relationships, this can become very ugly and hard to maintain. Instead, I propose that documentation allow the following format:

```
--! This is a description
--! of the entity.
--!
--! clk: comment
--! out_data : Description **port comment**
entity counter is
  port (
    clk: in std_logic; 
    out_data: out std_logic 
  );
end counter;
```

In this case, the documentation output should be the exact same where the parser looks for the name in the special comments above the entity and associates them with a port/generic when it sees the name of the port as the first part of the comment line string. This should apply to subprograms as well. For example, here is documentation I have for a VHDL procedure:

```
    -- send_raw_data_packets
    -- sends data packets to the driver formatted raw bytes. This will
    -- construct and push packets to the driver queue and the number of packets
    -- will depend on number of frames to send and number of bytes in each
    -- frame
    --
    -- net              : VUnit net global signal
    -- rnd              : RandomPType protected variable, init this outside prc
    -- num_frames       : Number of frames to send to UUT
    -- min_num_bytes    : min num of bytes
    -- max_num_bytes    : max num of bytes, this will randomly set number of
    --                    bytes in each frame constrained to this range
    -- num_bytes_vec    : integer_vector that will be used instead of random
    --                    num of bytes. If default or any 0's in vec, random
    --                    bytes used, else if this is non-zero and length1, all
    --                    frames will be the same num of bytes, else if length
    --                    > 1, this needs to be num_frames length and this is
    --                    how you can specify the number of bytes in each frame
    procedure send_raw_data_packets (
        signal net       : inout network_t;
        variable rnd     : inout RandomPType;
        num_frames       : positive       := 1;
        min_num_bytes    : positive       := 1;
        max_num_bytes    : positive       := 1;
        num_bytes_vec    : integer_vector := (0 => 0)
    );
```

As you can see, it would be really ugly if I had to put the port descriptions in-line with each procedure port.

The other thing that would be nice is to have a way of grouping signals in the port/generic. E.g. adding something like 

```
--! ~name~ 
--! <description for below signals/generics/parameters>
--! clk: comment
--!
--! ~name~
--! <description here for below signals/generics/parameters>
--! out_data : comment
```

qarlosalberto, 2020-11-16:
I agree with you, but I think that it's better this syntax:

```
--! This is a description
--! of the entity.
entity counter is
  port (
    --! Comment clk
    --! long description
    clk: in std_logic; 
    --! Comment out_data
    --! long description
    out_data: out std_logic 
  );
end counter;
```

What do you think?

smgl9, 2020-11-16:
> For the documentation of VHDL/V/SV, it would be nice if you can choose where this goes. Take the following example:
>
> ```
> --! This is a description
> --! of the entity.
> entity counter is
>   port ( [...]

Ok. I totally agree with you @GlenNicholls  But let's see some things to figure out how to solve the issue. 

In order to instantiate an entity I recommend to use the template generated by TerosHDL. It will not contain documentation.
The verbose inline documentation is a pain. For verbose documentation you can use html labels ( see example below ).

The problem I see in your proposal ( along with the likely headache with the parsers) is that the name of the port has to be maintained in the entity and the documentation to keep everything in the right place in documentation. This goes a bit against the philosophy of the tool.

Using html labels you can link in markdown and html the inline link with the actual documentation:

```
--! This is a description
--! of the entity.
--!
--! clk: comment   <a name="id1">name/nothing</a>
--! out_data : Description **port comment**   <a name="id2">name/nothing</a> 
entity counter is
  port (
    clk: in std_logic;              --! [name](#id1)
    out_data: out std_logic   --! [name](#id2)
  );
end counter;
```

> In this case, the documentation output should be the exact same where the parser looks for the name in the special comments above the entity and associates them with a port/generic when it sees the [...]

Yes. I also feel the need to group signals into ports. Documentation in subprograms is something we also have to add. 

In case html labels aren't enough, I think that the proposal from @qarlosalberto is neater and makes more sense. The generated template will remove the comments also in that case of use.

GlenNicholls, 2020-11-17:
Thanks for the help! Yes, @qarlosalberto suggestion would work and HTML links would also work!

qarlosalberto, 2021-01-08:
@GlenNicholls it has been implemented in the new TerosHDL version. Update it from VSCode market.

qarlosalberto, 2021-01-11:
It has been added to the new version: https://terostechnology.github.io/terosHDLdoc/features/documenter.html

el3ctrician, 2021-01-11:
that's great but the description parsing is broken in VHDL, any comments with the special tag before entity are simply ignored including the WaveDrom timing :( 

also it might be helpful to add a Doxygen style for the file header since the description can be parsed from the tag `@details` as in the following snippet
```
----------------------------------------------------------------------------
--! @file ff_re.vhd
--! @author el3ctrician
--! @brief Responsible of implementing a flipflop
--! @details a flip flop is a circuit that registers its input on the output only at the rising edge of the clock
--! @version 0.1
--! @date 2020-07-10
--!
--! @copyright 
--!                TerosHDL - GNU Public License
--! is free software: you can redistribute it and/or modify
--! it under the terms of the GNU General Public License as published by
--! the Free Software Foundation, either version 3 of the License, or
--! (at your option) any later version.
```
this could be parse into a description by combining the `@berif` and `@details` options and maybe ignore the other tags, or add optional support for them will be more than awesome !

also currently if   `@berif` and `@details` are used, for instance for a process, the result into generating hyperlinks .... so could at least this be avoided. 

I am willing to help if you guide me to where i can start digging in the code ;)

qarlosalberto, 2021-01-11:
Thank you for the feedback! Could you share your file? I'm testing a file and it works :/

Open an other issue for the Doxygen support, so we can discuss it :)

qarlosalberto, 2021-01-11:
@el3ctrician check the issue #66

el3ctrician, 2021-01-11:
after last update seems working with no issues now 👍

el3ctrician, 2021-01-11:
@qarlosalberto have a look at issue #67

closed 2021-02-02 by qarlosalberto

GlenNicholls, 2021-03-24:
Can you add information about HTML labels to the documentation?

qarlosalberto, 2021-03-24:
Could you add more info about? To parse HTML labels?

smgl9, 2021-03-24:
> Can you add information about HTML labels to the documentation?

Ok. I guess you are thinking about the suggestions made at the beginning of the issue. Good idea.

GlenNicholls, 2021-03-25:
@qarlosalberto @smgl9 I will open a new issue I think with a few issues related to the documenter I am seeing. But yes, I will include some code and screenshots in that.