TerosTechnology/vscode-terosHDL #47: [feature] Adding  Sublime Text SystemVerilog Package formatter

ereggiani, 2020-11-20:
Hi all,

thanks for sharing this project, it is really helpful.
It would be very interesting to have the possibility to easily format any SystemVerilog Code.

In the past I used the formatted from Sublime Text SystemVerilog Package and it works good:

https://github.com/TheClams/SystemVerilog
https://github.com/TheClams/SystemVerilog/blob/master/verilogutil/verilog_beautifier.py

It would be possible to integrate this formatter in TerosHDL for Visual Studio Code?

Thanks

ereggiani, 2020-11-22:
Hi,
thanks for your reply!

I was suggesting integrating the Sublime Text SystemVerilog Package because it supports a lot of SystemVerilog constructs and works very well.
Actually, I didn't try Verilog iStyle. Does it support SystemVerilog as well? Do you have any plan to extend the iStyle installation/configuration guide on the TerosHDL extension page?

qarlosalberto, 2020-11-24:
Yes, we should extend the documentation with a formatter guide. 
iStyle is installed with the IDE, you can use it; you don't need to install it. It supports SystemVerilog, although I don't know how well. 

We can also add what you suggest, it looks good.

qarlosalberto, 2021-03-26:
Added in: https://github.com/TerosTechnology/vscode-terosHDL/pull/100

closed 2021-03-26 by qarlosalberto