TerosTechnology/vscode-terosHDL #48: [Usage] Cannot configure linter properly

zlzhan, 2020-11-29:
[image]
Try to configure linter, but it didn't work

qarlosalberto, 2020-11-29:
Try with the xvlog folder path, delete "xvlog.exe"

zlzhan, 2020-11-30:
> Try with the xvlog folder path, delete "xvlog.exe"

I tried it, still to no avail.

zlzhan, 2020-11-30:
I also tried to close other Verilog related extensions, but they did not take effect.

qarlosalberto, 2020-11-30:
What is you Verilog file path?

zlzhan, 2020-11-30:
> What is your Verilog file path?

C:\Users\zahei\Desktop\project_3\source_v2

qarlosalberto, 2020-11-30:
1. Open a Verilog file.
2. Open: Help->Toggle Developer Tools->Console
3. Introduce an error in your Verilog file.

What is the output in the console?

zlzhan, 2020-11-30:
[image]
This is a screenshot, do you need text information?

qarlosalberto, 2020-11-30:
We have found the bug. We hope to release a new version tomorrow.

zlzhan, 2020-12-01:
Looking forward to the release.
By the way, the same problem occurred with the configuration of vhdl.

qarlosalberto, 2020-12-02:
There is a new version with the Vivado bug fixed (I hope).

zlzhan, 2020-12-02:
> There is a new version in the market with the Vivado bug fixed (I hope).

Thanks. To be honest, it was difficult to find a useful VHDL syntax check plugin before this. This really helps me a lot.

el3ctrician, 2020-12-03:
I just tested xvhdl with the new version and it is working now. I have it available system wide so i did not even configure the path. Thanks a lot.  Still not perfect due to [#54 ](https://github.com/TerosTechnology/vscode-terosHDL/issues/54) but we better discuss it there

closed 2020-12-10 by qarlosalberto