TerosTechnology/vscode-terosHDL #49: Other extended hosts keep shutting down.

pugdog93, 2020-11-30:
After installing vscode-terosHDL, extension keep failed when typing in settings.json

qarlosalberto, 2020-11-30:
Could you repeat it with Help->Toggle Developer Tools->Console open? What is the output?

qarlosalberto, 2020-11-30:
Could you share your settings.json?

pugdog93, 2020-12-01:
> Could you repeat it with Help->Toggle Developer Tools->Console open? What is the output?

I think there is extension collision between rainbow end extension and teroshdl

After disabling rainbow-end extension on workspace, there is no error

Error Log with when Rainbow end extension is enabled 
%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1
(function (exports, require, module, __filename, __dirname) { var Module=void 0!==Module?Module:{};!function(e,t){"function"==typeof define&&define.amd?define([],t):"object"==typeof exports?module.exports=t():window.TreeSitter=t()}(0,function(){var e,t={};for(e in Module)Module.hasOwnProperty(e)&&(t[e]=Module[e]);var n,r,o=[],s="./this.program",i=function(e,t){throw t},a=!1,_=!1;a="object"==typeof window,_="function"==typeof importScripts,n="object"==typeof process&&"object"==typeof process.versions&&"string"==typeof process.versions.node,r=!a&&!n&&!_;var u,l,d,c,f="";n?(f=_?require("path").dirname(f)+"/":__dirname+"/",u=function(e,t){return d||(d=require("fs")),c||(c=require("path")),e=c.normalize(e),d.readFileSync(e,t?null:"utf8")},l=function(e){var t=u(e,!0);return t.buffer||(t=new Uint8Array(t)),O(t.buffer),t},process.argv.length>1&&(s=process.argv[1].replace(/\\/g,"/")),o

[TypeError: Cannot read property 'caseSensitive' of undefined
	at Timeout.updateDecorations [as _onTimeout] (%USERPROFILE%\.vscode\extensions\jduponchelle.rainbow-end-0.7.7\out\extension.js:147:32)
	at listOnTimeout (internal/timers.js:531:17)
	at processTimers (internal/timers.js:475:7)]

P.S.

In VHDL, documentation doesn't work when "NULL" (all uppercase) in code. but null(all lower case) have no error. 

Error Log with NULL 

notificationsAlerts.ts:38 TypeError: Cannot read property 'replace' of null
	at Documenter._get_doc_ports (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:404:29)
	at Documenter._get_in_out_section (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:393:18)
	at Documenter._get_html_from_code (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:252:37)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async Documenter.get_html (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:78:20)

Thanks a lot for very useful extension !

qarlosalberto, 2020-12-01:
Thank you for the info! Could you share an example of the null bug?

pugdog93, 2020-12-01:
In VHDL for case statement

    WHEN OTHERS => NULL; 

    this code makes 

TypeError: Cannot read property 'replace' of null
	at Documenter._get_doc_ports (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:404:29)
	at Documenter._get_in_out_section (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:393:18)
	at Documenter._get_html_from_code (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:252:37)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async Documenter.get_html (%USERPROFILE%\.vscode\extensions\teros-technology.teroshdl-0.0.7\node_modules\jsteros\src\documenter\documenter.js:78:20)

fixing to 
    when others => null; 

makes no error 

when using formatter in terosHDL, this error doesn't appear since formatter makes null to lower case, but using other formatter, which makes uppercase, occurs error

qarlosalberto, 2020-12-01:
Are you sure about the error in NULL? I have found a bug when the port direction (IN,OUT) is uppercase. But I can't reproduce the bug in NULL. Could you share the complete code (deleting the confidential implementation)?

qarlosalberto, 2021-01-08:
Closed. Feel free to reopen.

closed 2021-01-08 by qarlosalberto