TerosTechnology/vscode-terosHDL #54: package and work package

el3ctrician, 2020-12-01:
hello,
is there is a way to add the xilinx libs to the project and my own custom packages that i use under work lib ?
the linter keeps showing errors about it 

thanks

qarlosalberto, 2020-12-01:
Yes, in the settings: Linter->VHDL/Verilog->Linter->xvhdl/xvlog->Arguments you can add custom arguments. E.g.:

```
--2008 --work work /my_rt_code/*.vhd
```

[image]

In the future it won't be manual.

el3ctrician, 2020-12-01:
thanks, i was able to set my work lib correctly using ghdl ( i am using it as linter since i plan to use vunit to test) but i still have issues with Xilinx libs. Please check the attached image
[image]

qarlosalberto, 2020-12-01:
For GHDL you need to compile the Xilinx libraries and add them to the configuration, e.g:

```
--std=08 -P/home/carlos/repo/ghdl-xilinx/unisim/v08 -P/home/carlos/repo/ghdl-xilinx/unimacro/v08
```

**linter->VHDL->Linter->ghdl ->Arguments**

PS: What plugin are you using to show the yellow port brackets?

el3ctrician, 2020-12-01:
Thanks !  i'll try and see if i can get it to work and i'll update the issue. 
the plugin is called `Bracket Pair Colorizer`

el3ctrician, 2020-12-02:
since i am using 93 standard my command is 
` --std=93c --workdir=C:\<absolute_path_to_pkgs_files> -PC:\<absolute_path_to_xilinx_compiled_libs>` but is till have no success. I used the script ` PS> <GHDL>\libraries\vendors\compile-xilinx-vivado.ps1 -All` to compile my xilinx libs but still no success. Launching the command from PowerShell gives no errors and seems to be working fine. I assume that work needs to be analyzed/elaborated  by GHDL at the beginning which produces a .o file in the "work" path that gets then "linked" when analyzing the actual source file. I think that a bit of work must be done to support external packages and libs and it depends on the linter it self. I also tried with xvhdl, which i have in my PATH, but xvhdl seems not working at all so no linting is done. As you can probably guess, i am using windows 10 home edition

el3ctrician, 2020-12-02:
ok, Xilinx compiled libs now seems to work correctly but unless i manually add a compiled version of my vhdl file containing the pkg, ghdl will not be able to find it

el3ctrician, 2020-12-02:
Not sure if it is the correct thing to do but i would add something to "lint" my work directory, with a proper option added for the linter settings under vscode ui, about here 
https://github.com/TerosTechnology/vscode-terosHDL/blob/62144f55884bc35b434a5d4013d65d58a8d09c10/src/lib/linter/linter.ts#L115
so it builds my work packages each time and add it to a proper dir ( maybe in a folder under workspace ? ) so il can be referenced to the linter ?
i am no ts expert but i think that it might work ... what do you think ?

qarlosalberto, 2020-12-02:
It's working for me in Linux:

```
--std=93c -P/home/carlos/repo/ghdl-xilinx/unisim/v93 -P/home/carlos/repo/ghdl-xilinx/unimacro/v93
```

What is the the output when you run in the cmd?:

```
ghdl -s --std=93c /pth/to/your/file.vhd -P/home/carlos/repo/ghdl-xilinx/unisim/v93 -P/home/carlos/repo/ghdl-xilinx/unimacro/v93
```
The linter only runs that command and parse the output.

There is a bug in the Vivado linter in Windows, I hope to release a new version this nigth.

el3ctrician, 2020-12-02:
yes as i was saying i managed to get the unisim and unimacro to work the main issue now is that all packages inside the work lib should be compiled in order for the `--workdir` option to be able to find them or at least that what i noticed from my test. 
said so running 
`ghdl -s --std=93c -PC:\active_projects\xilinx_lib --workdir=..\pkg  color_space_rotator.vhd` 
would result in :
```
color_space_rotator.vhd:5:10: unit "re_wi18_pkg" not found in library "work"
color_space_rotator.vhd:18:41: no declaration for "axi_lite_addr_width"
color_space_rotator.vhd:22:41: no declaration for "axi_lite_data_width"
color_space_rotator.vhd:23:42: no declaration for "axi_lite_data_width"
color_space_rotator.vhd:29:41: no declaration for "axi_lite_addr_width"
color_space_rotator.vhd:33:42: no declaration for "axi_lite_data_width"
color_space_rotator.vhd:62:21: entity 'color_space_rotator' was not analysed
```
i must be invoked as the following in order for the package to be recognized

`ghdl -s --std=93c -PC:\active_projects\xilinx_lib --workdir=..\pkg  ..\pkg\re_wi18_pkg.vhd .\color_space_rotator.vhd`

now that i thing about it it turns out to be much simpler of an issue to solve. If you can point me to where the command is constructed, i can work, slowly since am not a ts guy, on a crude implementation for adding two options and constructing the command correctly

PS: for xilinx compiled libs, no need to specify individual folders. It is enough to point once to the parent folder dir `/home/carlos/repo/ghdl-xilinx`

smgl9, 2020-12-02:
Thanks for your insight of the issue!
I think this issue is closely related with the project manager. Which is partially implemented but needs some revision to contemplate this kind of use cases and the integration with vscode. Maybe the best strategy is to have the project manager integrated first. Then we can add the necessary modifications if something is not working, but this issue should be solved  by then. The solution in on the way :)

qarlosalberto, 2020-12-02:
The command is here: https://github.com/TerosTechnology/colibri/blob/master/src/linter/ghdl.js#L28

The Vivado bug: #48

el3ctrician, 2020-12-03:
at this point xvhdl is working and hence no need to configure xilinx compiled lib in ghdl which is a huge plus. Still even when i add my work folder in xvhdl using the arguments option `--2008 --work work /my_rt_code/*.vhd` xvhdl stops working and i can't understand why actually. Anyway the issue seems known as know as suggested by @smgl9 so if i can help in some way please let me know. 
@qarlosalberto thanks for pointing out the linter command but at this point i think it makes more sense to wait for the project management feature before touching the code to avoid reworking ... what do you think ?

el3ctrician, 2020-12-04:
Waiting for a final solution for now i found this work around 

`-work work <absolute path to single file where the package is defined>`

i haven't tested it with multiple files but should be simply absolute path followed by space.

If you put all you files in one folder no issues are found

qarlosalberto, 2020-12-04:
Thank you for the solution! Yes, I think that it's better to wait, with the project manager it will be easier.

el3ctrician, 2021-01-11:
just a quick update, seems that in the last update the issue has been fixed so all packages are recognized automatically, i guess by looking at the opened files, hopefully it is much easier now with XVHDL. Thanks for the great work guys and please keep it up

qarlosalberto, 2021-01-11:
Thank you for the feeback! In the last update we have not added anything related to this. But we are working on the next update.

el3ctrician, 2021-01-11:
let me know if I can help in anyway then

qarlosalberto, 2021-04-15:
It has been added with RustHDL.

closed 2021-04-15 by qarlosalberto