TerosTechnology/vscode-terosHDL #63: References to multiple signal declaration not recognized

avlixa, 2021-01-09:
When referencing signals with a multiple declaration is not reconized, neither with Go to definition option (F12)
This is an example of declaration
[image]

The signal cart_a_s is well recognized:
[image]

The signal cart_d_s is not recognized:
[image]

My source code is a modification of this one [https://github.com/yomboprime/VideoPac-ZXDOS/blob/master/src/board/ZX_Uno_xs6/ZX_Uno_xs6_vp.vhd](url) with some changes, but the issue is already in that one.

qarlosalberto, 2021-01-11:
Thank you for the feedback! It will be fixed in the next release.

qarlosalberto, 2021-01-11:
We have released a new version with some small bugs fixed. Can you check if it's fixed?

closed 2021-02-02 by qarlosalberto