TerosTechnology/vscode-terosHDL #65: Verible does not work on SV files

ila-embsys, 2021-01-11:
TerosHDL 0.1.0

If file extension ends with _'v'_ -- problems tab shows lines with _'TerosHDL:verible'_
If the same file is renamed to have extension as _'sv'_ -- no lines with _'TerosHDL:verible'_ are shown anymore.

Direct call _verible-verilog-lint_ shows the same output on _'v'_ files and _'sv'_ files as problems tab for _'v'_ files. 

Expected that _Verible_ should work with SystemVerilog files as well as Verilog

qarlosalberto, 2021-01-11:
Thank you for the feedback! I have found the bug. I will be fixed in the next release.

qarlosalberto, 2021-01-11:
We have released a new version with some small bugs fixed. Can you check if it's fixed?

ila-embsys, 2021-01-12:
Yes, it fixed. SystemVerilog files now show problems generated by verible. Thanks!

closed 2021-01-12 by qarlosalberto