TerosTechnology/vscode-terosHDL #66: Wavedrom & Bitfield Disabled in 0.1.0

dra5546, 2021-01-11:
Up until the latest release, Wavedrom Bitfield were working as expected. In the latest release neither seem to show up in the doc viewer or the HTML doc I generated. 

Note: Tried out the State machine viewer for ~ 10 state machines and it worked beautifully

qarlosalberto, 2021-01-11:
Can you share a file where it fails?

dra5546, 2021-01-11:
I'm unable to share source code but I'm certainly willing to share the snippet of wavedrom I'm using. I've tried to place the snippet in several sections of code with and without state machines and it behaves the same. Below is the code that was generating a wavedrom diagram in the previous version and now is not.

--! **WaveDrom** UART TX Diagram:
--!{ signal: [
--!  {name: 'i_TX_Byte',         wave: '.27...........8..........x', data:['TXData1','TXData2']},
--!  {name: 'i_TX_DV',         wave: 'l.pl..........pl..........'},
--!  {name: 'o_TX_Serial',         wave: '.274444444459x74444444459x', data:['Start','Bit0','Bit1','Bit2','Bit3','Bit4','Bit5','Bit6','Bit7','Parity','Stop','Start','Bit0','Bit1','Bit2','Bit3','Bit4','Bit5','Bit6','Bit7','Parity','Stop']},
--!   {name: 'Serial Clock',        wave: 'l.P......................l'},
--!  ['Event',
--!   {name: 'o_TX_Done',     wave: 'l............pl..........p'},
--!   {name: 'o_TX_Active',   wave: 'l.h..........lh..........l'}
--!  ]
--!]
--!}

smgl9, 2021-01-11:
I've been doing some test. It seems that wavedrom stops working as usual and is something we have to fix. 
As a workaround for now try to put the wavedrom snippet before the library declaration and without common comments after the snippet

qarlosalberto, 2021-01-11:
We have released a new version with some small bugs fixed. Can you check if it's fixed?

dra5546, 2021-01-11:
It works with a minor difference than before. In the previous version I could put the Wavedrom code above my library declarations (i.e. in my code header) and it would generate it correctly. In this case it seems like I must put the wavedrom below my library declaration.

Thank you for the fix

qarlosalberto, 2021-01-11:
It is because in the future it will be possible to document libraries.
The VHDL parser has been completely changed due to the state machines.
I try that the changes do not affect the user, but as the IDE is under development, sometimes it is not possible. I hope the changes are minimal.

qarlosalberto, 2021-01-11:
Feel free to suggest changes to how comments are extracted.

closed 2021-01-11 by qarlosalberto