TerosTechnology/vscode-terosHDL #67: Better "Than" Doxygen Support

el3ctrician, 2021-01-11:
First of all i would like to thank all of you guys for the awesome project. I think that, at least me, desperately needed some modern tools for Digital Designer as my older colleges tend to used notepad++ or vim or something else. You really did a great job so far and i would love to help see it improves and get better overtime. Following are some of my thoughts regarding the documentation tool "Documenter" that could be the best opensource tool available in my opinion. What i love most about it that it is clutter free as it gives useful information that could guide someone who reads the code for the first time while keeping the developer effort to minimum. In the past i was relying on Doxygen to get the job done but Doxygen generates a lot of useless stuff and repeated documentation when a small but efficient documentation is needed. Sure thing that it can be customized but it takes huge effort to do so specially for VHDL due to the lack of dedicated VHDL "mentality" since it treats VHDL some times as a generic OOP which is completly a mess.   

That said some of the features can be "borrowed" from Doxygen structure while adding support for new specific structure for digital design. 

For instance the command concept of Doxygen, more infos [here](https://www.doxygen.nl/manual/commands.html)  , is very powerful. For instance usually all my file have a header similar to the following snippet : 

```
----------------------------------------------------------------------------
--! @file ff_re.vhd
--! @author el3ctrician
--! @brief Responsible of implementing a flipflop
--! @details a flip flop is a circuit that registers its input on the output only at the rising edge of the clock
--! @version 0.1
--! @date 2020-07-10
--!
--! @copyright 
--!                TerosHDL - GNU Public License
--! is free software: you can redistribute it and/or modify
--! it under the terms of the GNU General Public License as published by
--! the Free Software Foundation, either version 3 of the License, or
--! (at your option) any later version.
``
`currently this gets ignores by the documentation tool and nothing is parsed. So adding both the  `@berif` and `@details` commands would be a good start since it is also used sometimes to describe a process. If it used to describe  a process the current behavior is that i found a generated link to github.com/berif or details in my pdf/html. 
We might also consider to support the rest of them and let them be optional in dedicated signals that appears in the output if it is found (version, copyright ..etc.) otherwise it is the same as now.

This could be a great starting point after that we can add a support for grouping signal to one bus as per the following snippet : 

``` vhdl
entity video_core is 
  port (
    --! core clock, 100 Mhz
    clk : in std_logic;
    --! asynchronous active low reset
    reset_n : in std_logic;

   --! @startbus axi_lite_slave 
    s_axi_awaddr  : in std_logic_vector(AXI_LITE_ADDR_WIDTH - 1 downto 0);
    s_axi_awprot  : in std_logic_vector(2 downto 0);
    s_axi_awvalid : in std_logic;
    s_axi_awready : out std_logic;
    s_axi_wdata   : in std_logic_vector(AXI_LITE_DATA_WIDTH - 1 downto 0);
    s_axi_wstrb   : in std_logic_vector((AXI_LITE_DATA_WIDTH/8) - 1 downto 0);
    s_axi_wvalid  : in std_logic;
    s_axi_wready  : out std_logic;
    s_axi_bresp   : out std_logic_vector(1 downto 0);
    s_axi_bvalid  : out std_logic;
    s_axi_bready  : in std_logic;
    s_axi_araddr  : in std_logic_vector(AXI_LITE_ADDR_WIDTH - 1 downto 0);
    s_axi_arprot  : in std_logic_vector(2 downto 0);
    s_axi_arvalid : in std_logic;
    s_axi_arready : out std_logic;
    s_axi_rdata   : out std_logic_vector(AXI_LITE_DATA_WIDTH - 1 downto 0);
    s_axi_rresp   : out std_logic_vector(1 downto 0);
    s_axi_rvalid  : out std_logic;
    s_axi_rready  : in std_logic;
    --! @endbus axi_lite_slave 

    --! @startbus axis_video_in
    video_in_tdata  : in  std_logic_vector(23 downto 0);
    video_in_tlast  : in  std_logic;
    video_in_tuser  : in  std_logic_vector(0 downto 0);
    video_in_tvalid : in  std_logic;
    video_in_tready : out std_logic;
   --! @endbus axis_video_in

    --! @startbus axis_video_out
    video_out_tdata  : out std_logic_vector(23 downto 0);
    video_out_tlast  : out std_logic;
    video_out_tuser  : out std_logic_vector(0 downto 0);
    video_out_tvalid : out std_logic;
    video_out_tready : in  std_logic
   --! @startbus axis_video_out
  );
end video_core;

--! @busdescription axi_lite_slave  An AXI4-Lite compliant slave interface to configure the core registers, the register width is 4 bytes
--! @busdescription axis_video_in An AXI4-Stream compliant slave interface to receive the video frame
--! @busdescription axis_video_in An AXI4-Stream compliant master interface to send the video frame out

``` 

I think that this could improve the readability and the quality of the documentation a lot.

Then maybe, in a very doxygen style, also a way to reference other components in the code and so a global way for documenting the project as all which creates a single pdf/HTML for all the entities and allows to follow hyperlinks between various components which leads to a better documentation for the overall project. A cherry on top at this matter it to support for instance a link to .md file as main page (similar to doxygen ` @mainpage `

Also wavedrom and bitfield are very powerful tools but i would like to see wavedrom supported more than once. This is useful for complex designs where in some cases there are multiple processes inside an entity and so i was thinking maybe we can keep them supported for as once for each process ?

I would love to dedicate some of my spare time to work on some of these features if you can supervise and guide me through the code.

Thanks !

smgl9, 2021-01-11:
> First of all i would like to thank all of you guys for the awesome project. I think that, at least me, desperately needed some modern tools for Digital Designer as my older colleges tend to used [...]
Thank you @el3ctrician ! We are really pleased with your comments :) And thanks for your feedback. 

We also used Doxygen in the past. I agree with you that we can take some ideas from it.

> That said some of the features can be "borrowed" from Doxygen structure while adding support for new specific structure for digital design.
>
> For instance the command concept of Doxygen, more infos [...]

I totally agree with you. There is no doubt that is something we need. I like your idea about grouping ports but I'm still thinking how to implement it neatly. 

The terosHDL backend is in charge of parsing and generating the documentation (colibri) I can point you to the files in case you want to take a look and share your ideas or implementations.  For VHDL https://github.com/TerosTechnology/colibri/blob/master/src/parser/ts_vhdl_parser.js The oputput is a JSON with all the information to generate the documentation.

> Then maybe, in a very doxygen style, also a way to reference other components in the code and so a global way for documenting the project as all which creates a single pdf/HTML for all the entities [...]
I'm a bit confused, not sure if you are aware if this functionality: https://terostechnology.github.io/terosHDLdoc/features/dependencies_viewer.html or you are thinking about something different... Browse to the bottom of the page, you can see a gif animation showing the generated documentation.

> Also wavedrom and bitfield are very powerful tools but i would like to see wavedrom supported more than once. This is useful for complex designs where in some cases there are multiple processes inside an entity and so i was thinking maybe we can keep them supported for as once for each process ?
> 

I guess that it could be added @qarlosalberto .

> I would love to dedicate some of my spare time to work on some of these features if you can supervise and guide me through the code.
> 
> Thanks !

For sure. We would be more than happy! In the last weeks the changes in the code have been huge.  Some interfaces (like parsers) have been rebuilt several times in the last months and remain undocumented. We are now doing some housekeeping and trying to add as many test as possible, cleaning some files, etc. I guess It would be easier to everybody in some days. But anyway, tell us what part are you willing to improve. Hopefully in some days we will document and publish the parsers interface and we can discuss about it :) And there is a lot of room for improvements and collaborations for sure.

Thanks!

el3ctrician, 2021-01-12:
I am very  happy that you found my argument interesting and you are willing to dedicate some thoughts to it that is fantastic 💯 

> I totally agree with you. There is no doubt that is something we need. I like your idea about grouping ports but I'm still thinking how to implement it neatly.
>
> The terosHDL backend is in charge of [...]

that's great! so i had a look at the file and i think that  the best way at this point to add explicit support for custom commands and treat them as language grammar, as process or architecture, but i assume that would be more complicated. Another way is to extended the current functions, look at  the one down for signal grouping for example, which the new supported commands but i think that could be less neat and more time consuming to maintain 

https://github.com/TerosTechnology/colibri/blob/4d3ca8998f5bbc242396c1ddc52be9d418f3f204/src/parser/ts_vhdl_parser.js#L218

I am not very familiar with web-tree-sitter but if there is a way to extract all the comments from a file, filter them by the special comment we might then parse all special comments into this way i'll dig deeper in the following days and keep you updated on this 

> I'm a bit confused, not sure if you are aware if this functionality: https://terostechnology.github.io/terosHDLdoc/features/dependencies_viewer.html or you are thinking about something different... Browse to the bottom of the page, you can see a gif animation showing the generated documentation.

yes i am aware about it but this generates a "homepage" which then can let you move between single files documentations. If you have both entity1 and entity2 where entity1 adds entity2 as instance to it's design you can still see entity2 in the documentation of entity1 but without a hyperlink to go directly to it. Also there is no way to add custom home page or description. This is very important since a lot of modern design are done in the schematics entry (BD in Xilinx) in order to use a soft/hard microcontroller and also vendor specific IP. This means that usually no actual TOP-Level is written manually in VHDL so it is useful to have a way to document also what happens in other parts of the project that could be helpful to understand the context of each component. So mainly if this feature could be extended with custom main page or custom pages in general would be awesome. Also if it gets awareness about the instance available in the documentation so can point to them with hyperlink would be great!

> For sure. We would be more than happy! In the last weeks the changes in the code have been huge. Some interfaces (like parsers) have been rebuilt several times in the last months and remain [...]
 
Great, so maybe i can wait for the public release and then we can come back and discuss about how we can start working on that! 

Thanks !!

qarlosalberto, 2021-01-12:
The parser output for VHDL and Verilog is a dictionary with the following structure:

https://github.com/TerosTechnology/colibri/blob/master/test/examples/vhdl/example_0.json

The extracted comments are in the "description" field. So I think that the best way to implement the Doxygen commands and the port port grouping is to do a post-processing step over the dictionary.

E.g. if you want to extract the command @autor:

1. Get the dictionary.
2. Parse the command @autor in the field entity->description.
3. Add the field autor to the dictionary. Delete @autor command in the field entity->description.

The next step is to show the commands in the HTML page. But it's other step. You can start to creating a post-processing function:
- Input: dictionary
- Output: post-processed dictionary

For these tests you can create a simple javascript file with the function, to read a .json example and to print the post-processed dictionary. Other examples:

https://github.com/TerosTechnology/colibri/tree/master/test/examples/vhdl

https://github.com/TerosTechnology/colibri/tree/master/test/examples/verilog

el3ctrician, 2021-01-12:
that seems straight forward and i can work with that, could you give me some tips on how to setup a development environment ? should i pull the latest sources from master which is the updated ones ? 

thanks !

qarlosalberto, 2021-01-12:
You can use the master branch. For to read a .json in a dictionary:

```
    let json = fs.readFileSync(__dirname + '/examples/verilog/example_' + x + '.json', 'utf8');
    let dic = JSON.parse(json);
```

( https://github.com/TerosTechnology/colibri/blob/master/test/parser.js#L31 )

You don't really need to use the rest of the components, you can create a js file and later to integrate it. 

If you want to develop in colibri:

1. git clone https://github.com/TerosTechnology/colibri.git
2. cd colibri
3. npm install (it will download the dependencies)
4. Import colibri in VSCode
5. You can debug your js file with VSCode debugger.

[image]

el3ctrician, 2021-01-12:
@qarlosalberto thanks alot. I already started and i will move the discussion on the dev to gitter so i'll be reaching you there if it is ok with you.