TerosTechnology/vscode-terosHDL #68: extension crashes extension host

lepommosaure, 2021-01-25:
First of all a big thank you to the dev team, this extension looks very promising, especially the automated documentation generation!

On my machine I have a very recurring crash when I use terosHDL: "Extension host terminated unexpectedly"

The DevTools give me the following error:
```
c:\Users\user_name\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1
(function (exports, require, module, __filename, __dirname) { var Module=void 0!==Module?Module:{};!function(e,t){"function"==typeof define&&define.amd?define([],t):"object"==typeof exports?module.exports=t():window.TreeSitter=t()}(0,function(){var e,t={};for(e in Module)Module.hasOwnProperty(e)&&(t[e]=Module[e]);var n,r,o=[],s="./this.program",i=function(e,t){throw t},a=!1,_=!1;a="object"==typeof window,_="function"==typeof importScripts,n="object"==typeof process&&"object"==typeof process.versions&&"string"==typeof process.versions.node,r=!a&&!n&&!_;var u,l,d,c,f="";n?(f=_?require("path").dirname(f)+"/":__dirname+"/",u=function(e,t){return d||(d=require("fs")),c||(c=require("path")),e=c.normalize(e),d.readFileSync(e,t?null:"utf8")},l=function(e){var t=u(e,!0);return t.buffer||(t=new Uint8Array(t)),O(t.buffer),t},process.argv.length>1&&(s=process.argv[1].replace(/\\/g,"/"))

[RuntimeError: abort(Error: spawn c:\Users\user_name\Watt\project_name\git ENOENT). Build with -s ASSERTIONS=1 for more info.
	at process.we (c:\Users\user_name\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1:11401)
	at process.emit (events.js:228:7)
	at processPromiseRejections (internal/process/promises.js:201:33)
	at processTicksAndRejections (internal/process/task_queues.js:95:32)]
```
I'm using **Windows 10**
Could it be that it is looking for a **git** directory? Sadly I am not using git but subversion... :)

Thanks a lot!

qarlosalberto, 2021-01-25:
Could you give us more info?

When do it happen? With the Documenter, state machine viewer...? Are you using VHDL, Verilog or SV?

lepommosaure, 2021-01-25:
Now that's a quick answer! :)

1. I am using VHDL
2. exporting markdown in "module documentation" seems to do the trick and launch the "crash loop"
2. Once the crash loop has started, restarting VSCode or extension host does not stop the extension host from crashing immediately again
3. The only fix I found seems to be: disabling TerosHDL, restarting extension host, enabling TerosHDL again

lepommosaure, 2021-01-25:
Adding a folder to a new workspace crashes too... Could it be some kind of folder parsing triggered on some events?

qarlosalberto, 2021-01-25:
- Have you the same error when you export to HTML? And to diagram?
- Where are you saving the file? Can you share your path? Spaces or special symbols?

Thank you :)

qarlosalberto, 2021-01-25:
Is it created the file?

lepommosaure, 2021-01-25:
I suppose the trigger for the error is not precisely known yet because I just tried to export and it worked perfectly fine...

When it works OK:
- HTML is fine (no error)
- Markdown OK too
- No spaces nor special symbols in path (only A-Z 0-9 a-z and _ )
- SVG diagrams export also OK

then I used "export PDF", with the following issue:
_Currently this feature is stable in Linux. Report your problem._
and then the "extension host terminated unexpectedly" appears again, with the following detailed error:

```
c:\Users\user_name1\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1
(function (exports, require, module, __filename, __dirname) { var Module=void 0!==Module?Module:{};!function(e,t){"function"==typeof define&&define.amd?define([],t):"object"==typeof exports?module.exports=t():window.TreeSitter=t()}(0,function(){var e,t={};for(e in Module)Module.hasOwnProperty(e)&&(t[e]=Module[e]);var n,r,o=[],s="./this.program",i=function(e,t){throw t},a=!1,_=!1;a="object"==typeof window,_="function"==typeof importScripts,n="object"==typeof process&&"object"==typeof process.versions&&"string"==typeof process.versions.node,r=!a&&!n&&!_;var u,l,d,c,f="";n?(f=_?require("path").dirname(f)+"/":__dirname+"/",u=function(e,t){return d||(d=require("fs")),c||(c=require("path")),e=c.normalize(e),d.readFileSync(e,t?null:"utf8")},l=function(e){var t=u(e,!0);return t.buffer||(t=new Uint8Array(t)),O(t.buffer),t},process.argv.length>1&&(s=process.argv[1].replace(/\\/g,"/"))

[Error: spawn c:\Users\user_name1\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\phantomjs-prebuilt\lib\phantom\bin\phantomjs ENOENT
	at Process.ChildProcess._handle.onexit (internal/child_process.js:264:19)
	at onErrorNT (internal/child_process.js:456:16)
	at processTicksAndRejections (internal/process/task_queues.js:81:21)] {
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\phantomjs-prebuilt\\lib\\phantom\\bin\\phantomjs',
  path: 'c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\phantomjs-prebuilt\\lib\\phantom\\bin\\phantomjs',
  spawnargs: [
    'c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\markdown-pdf\\phantom\\render.js',
    'C:\\Users\\user_name1\\AppData\\Local\\Temp\\tmp-14908VMyyUy0v4PU8.html',
    'C:\\Users\\user_name1\\AppData\\Local\\Temp\\tmp-14908y70fxcprS67K.pdf',
    'C:\\Users\\user_name1\\Desktop\\Apps\\Microsoft VS Code',
    'c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\markdown-pdf\\runnings.js',
    'c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\src\\documenter\\custom.css',
    'c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\markdown-pdf\\css\\highlight.css',
    'A4',
    'portrait',
    '2cm',
    0,
    10000
  ],
  cmd: 'c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\phantomjs-prebuilt\\lib\\phantom\\bin\\phantomjs c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\markdown-pdf\\phantom\\render.js C:\\Users\\user_name1\\AppData\\Local\\Temp\\tmp-14908VMyyUy0v4PU8.html C:\\Users\\user_name1\\AppData\\Local\\Temp\\tmp-14908y70fxcprS67K.pdf C:\\Users\\user_name1\\Desktop\\Apps\\Microsoft VS Code c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\markdown-pdf\\runnings.js c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\src\\documenter\\custom.css c:\\Users\\user_name1\\.vscode\\extensions\\teros-technology.teroshdl-0.1.3\\node_modules\\jsteros\\node_modules\\markdown-pdf\\css\\highlight.css A4 portrait 2cm 0 10000'
}
```

And once that error appeared one time, I then have this error each time I "Restart Extension host":
```

c:\Users\user_name\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1
(function (exports, require, module, __filename, __dirname) { var Module=void 0!==Module?Module:{};!function(e,t){"function"==typeof define&&define.amd?define([],t):"object"==typeof exports?module.exports=t():window.TreeSitter=t()}(0,function(){var e,t={};for(e in Module)Module.hasOwnProperty(e)&&(t[e]=Module[e]);var n,r,o=[],s="./this.program",i=function(e,t){throw t},a=!1,_=!1;a="object"==typeof window,_="function"==typeof importScripts,n="object"==typeof process&&"object"==typeof process.versions&&"string"==typeof process.versions.node,r=!a&&!n&&!_;var u,l,d,c,f="";n?(f=_?require("path").dirname(f)+"/":__dirname+"/",u=function(e,t){return d||(d=require("fs")),c||(c=require("path")),e=c.normalize(e),d.readFileSync(e,t?null:"utf8")},l=function(e){var t=u(e,!0);return t.buffer||(t=new Uint8Array(t)),O(t.buffer),t},process.argv.length>1&&(s=process.argv[1].replace(/\\/g,"/"))

[RuntimeError: abort(Error: spawn c:\Users\user_name\Watt\PROJ\FPGA_NAME\branches\inc2_debug\git ENOENT). Build with -s ASSERTIONS=1 for more info.
	at process.we (c:\Users\user_name\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1:11401)
	at process.emit (events.js:228:7)
	at processPromiseRejections (internal/process/promises.js:201:33)
	at processTicksAndRejections (internal/process/task_queues.js:95:32)]
```

Hope this helps...

lepommosaure, 2021-01-25:
When the error does not appear, the files are created successfully

qarlosalberto, 2021-01-25:
Thank you @lepommosaure !! Then when you don't export to PDF, everything works fine, right? 
It seems a problem with the function that exports to PDF... :/

lepommosaure, 2021-01-25:
Well I don't know, but at some point something checks whether a **git** folder exists or not, and fails if not, that seem to be an independent problem, triggered by the first failure of "export to PDF"

I'll tell you there if another the error happens again without exporting to PDF

qarlosalberto, 2021-01-25:
Thank you! I don't use Windows, so it's difficult for me test it.

koog1000, 2021-01-25:
TerosHDL also crashes my VSCode extension host. This happens right when I open VSCode.

qarlosalberto, 2021-01-25:
> I suppose the trigger for the error is not precisely known yet because I just tried to export and it worked perfectly fine...
>
> When it works OK:
>
>     * HTML is fine (no error)
>
>     * Markdown OK [...]

@koog1000 could you try the same? Have you the same problem?

koog1000, 2021-01-25:
I didn't even try to do anything with Teros. I just installed the extension and then opened some HDL files. When I reloaded the window it crashed, and continued crashing on each window reload.

I disabled the extension and reloaded VSCode again and now it is not crashing when I re-enable the extension.
If it happens again I'll try to get more info and possibly make a new issue depending on what I find out.

lepommosaure, 2021-01-25:
@koog1000, are you using git ? do you have a git folder at the root of your workspace?

koog1000, 2021-01-25:
Yes I am using git and the root of the workspace is the root of a git repository. That repository also has a couple submodules, not sure if that is relevant.

qarlosalberto, 2021-01-25:
I think it is not related to GIT. TerosHDL doesn't use git in anywhere. 

@koog1000 are you using Windows?

@lepommosaure tell me any error, please :)

koog1000, 2021-01-25:
@qarlosalberto I'm using Ubuntu 18.04.
I've had it error on me a few more times now. The behavior is strange... there will be no error, everything is working (or at least I it isn't crashed) and then I will reload the window and it will crash the extension host, and subsequent reloads crash the host until I disable the Teros extension. 

I'm still trying to narrow down what causes the initial crash to occur though.

qarlosalberto, 2021-01-25:
What are you using? Documenter, state machine viewer..?

smgl9, 2021-01-25:
> @qarlosalberto I'm using Ubuntu 18.04.
> I've had it error on me a few more times now. The behavior is strange... there will be no error, everything is working (or at least I it isn't crashed) and [...]

Ok... I'm using Ubuntu 18.04 as well, but I can't reproduce the issue. I have some questions: 
- What do you mean by "reloading the window"? 
- May you be activating the debugger (F5)?
- What are the actions that precede the reloading?

Thanks in advance!

smgl9, 2021-01-25:
@koog1000 @lepommosaure 

Could you tell us what other extensions you have installed? Maybe there are conflicts between extensions. 
Or at least, could you try TerosHDL extension alone?

lepommosaure, 2021-01-26:
I tried to disable every extension and the error appeared when doing "module documentation" >"exporting markdown", and trying to save the .md to a directory with VHDL sources

`ERR Extension host terminated unexpectedly. The following extensions were running: vscode.microsoft-authentication, hbenl.vscode-test-explorer, vscode.debug-auto-launch, vscode.emmet, vscode.git, vscode.merge-conflict, vscode.search-result, teros-technology.teroshdl, vscode.github, ms-vscode.js-debug, vscode.github-authentication`

the .md file was **not** created, and the following error happened:
```
Extension Host
localProcessExtensionHost.ts:258 c:\Users\user\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1
(function (exports, require, module, __filename, __dirname) { var Module=void 0!==Module?Module:{};!function(e,t){"function"==typeof define&&define.amd?define([],t):"object"==typeof exports?module.exports=t():window.TreeSitter=t()}(0,function(){var e,t={};for(e in Module)Module.hasOwnProperty(e)&&(t[e]=Module[e]);var n,r,o=[],s="./this.program",i=function(e,t){throw t},a=!1,_=!1;a="object"==typeof window,_="function"==typeof importScripts,n="object"==typeof process&&"object"==typeof process.versions&&"string"==typeof process.versions.node,r=!a&&!n&&!_;var u,l,d,c,f="";n?(f=_?require("path").dirname(f)+"/":__dirname+"/",u=function(e,t){return d||(d=require("fs")),c||(c=require("path")),e=c.normalize(e),d.readFileSync(e,t?null:"utf8")},l=function(e){var t=u(e,!0);return t.buffer||(t=new Uint8Array(t)),O(t.buffer),t},process.argv.length>1&&(s=process.argv[1].replace(/\\/g,"/"))

[RuntimeError: abort(Error: ENOENT: no such file or directory, open 'C:/Users/user/Watt/PROJ/FPGA_AAA_AAA/branches/inc2_debug/Libraries/sync_controller/HDL\stm_sync_ocu_dpll_00.svg
'). Build with -s ASSERTIONS=1 for more info.
	at process.we (c:\Users\user\.vscode\extensions\teros-technology.teroshdl-0.1.3\node_modules\jsteros\node_modules\web-tree-sitter\tree-sitter.js:1:11401)
	at process.emit (events.js:228:7)
	at processPromiseRejections (internal/process/promises.js:201:33)
	at processTicksAndRejections (internal/process/task_queues.js:95:32)]
```

You're right @qarlosalberto , no mention of git anywhere this time, sorry for insisting :)

qarlosalberto, 2021-01-26:
@lepommosaure you have the error when the IP contains a state machine, right?

qarlosalberto, 2021-01-26:
What is your complete file path? C:/Users/user/Watt/PROJ/FPGA_AAA_AAA/branches/inc2_debug/Libraries/sync_controller/HDL/stm_sync_ocu_dpll.vhd?

lepommosaure, 2021-01-26:
You're right, when the IP contains a state machine and that the checkbox "include state machines" is checked, markdown (and probably every other filetype) save triggers the error

The filepath is very similar to the one you wrote, I just changed a few letters out of misplaced paranoia :)

qarlosalberto, 2021-01-26:
I see in your log an error in the path. Did you introduce the error with your modification or is it so? 

**[RuntimeError: abort(Error: ENOENT: no such file or directory, open 'C:/Users/user/Watt/PROJ/FPGA_AAA_AAA/branches/inc2_debug/Libraries/sync_controller/HDL\stm_sync_ocu_dpll_00.svg
'). Build with -s ASSERTIONS=1 for more info.**

**C:/Users/user/Watt/PROJ/FPGA_AAA_AAA/branches/inc2_debug/Libraries/sync_controller/HDL\stm_sync_ocu_dpll_00.svg**

There is an error with the last "\\". It's a Linux character.

lepommosaure, 2021-01-26:
I did not introduce this error

qarlosalberto, 2021-01-26:
Could you send an email to terostechnology@gmail.com? I would like to share with you a test version.

smgl9, 2021-01-29:
@lepommosaure 

We've been able to reproduce the error in a windows machine. Hope to have a fix soon!

jamxe, 2021-02-03:
same problem, I turned all VSCode extensions and turned them on one by one and found that terosHDL caused the crash `Extension Host Terminated unexpectedly` whenever I want to create a new folder in the sidebar of VSCode.

qarlosalberto, 2021-02-03:
@jamxe are you using Windows or Linux? Do it work until you add a folder?

jamxe, 2021-02-04:
> @jamxe are you using Windows or Linux? Do it work until you add a folder?

I am using Windows 10 pro 20h2. It works after click `Restart Extension Host`.
[image]

qarlosalberto, 2021-02-04:
Could you share your VHDL/Verilog extensions? Maybe it's a problem with other extensions. Is everything working correctly now?

jamxe, 2021-02-05:
> Could you share your VHDL/Verilog extensions? Maybe it's a problem with other extensions. Is everything working correctly now?

I turned off other extensions except TerosHDL and found that there is a conflit when the other extension named [koroFileHeader](https://marketplace.visualstudio.com/items?itemName=OBKoro1.korofileheader) was enabled. The latter extension was used for inserting user snippets when file was created.

jamxe, 2021-02-05:
> I turned off other extensions except TerosHDL and found that there is a conflit when the other extension named [...]

My settings:

```json
"fileheader.customMade": {
    "Version": "1.0",
    "Date": "Do not edit",
    "LastEditTime": "Do not edit",
    "FilePath": "Do not edit"
    "Description": ""
    },
    "fileheader.cursorMode": { 
        "description":"",
        "param": "",
        "return": ""
    },
    "fileheader.configObj": {
        "atSymbol": [ "", "@"], 
        "specialOptions":{
            "Date": "Created on",
            "LastEditTime": "Last Modified"
          },
        "autoAdd": true, 
        "prohibitAutoAdd": [ "json", "md","h" ],
        "wideSame": true,
        "wideNum": 13,
        "language": {
            "java": {
                "head": "/$$",
                "middle": " $ ",
                "end": " $/"
            },
            "verilog": {
                "head": "/*",
                "middle": " * ",
                "end": "**/"
            },
           "h/hpp/cpp": {
              "head": "/*** ",
              "middle": " * ",
              "end": " */"
            },
            "blade.php":{
              "head": "<!--",
              "middle": " * @",
              "end": "-->",
            }
        },
    }
```

qgzln, 2021-02-28:
when i use open dependencies viewer add source then 
[image]

qarlosalberto, 2021-02-28:
@qgzln is Python3 installed in your system and added to the system path?

qgzln, 2021-03-01:
> @qgzln is Python3 installed in your system and added to the system path?
I am sure Python3 is  installed and added to the system path，Except for this button, other buttons are fine。sometimes，the statemachine viewer could not find the statemachine

qarlosalberto, 2021-03-01:
Follow these steps:

1. Configure your python3 path in Teroshdl settings.
2. Install VUnit: sudo pip3 install vunit_hdl

smgl9, 2021-03-11:
@lepommosaure The issue saving documentation should have been resolved in v0.1.4

jamxe, 2021-03-17:
> @lepommosaure The issue saving documentation should have been resolved in v0.1.4

Thanks! It works fine:)

qarlosalberto, 2021-06-01:
@heartacker open other issue with your problem and description.

closed 2021-06-01 by qarlosalberto