TerosTechnology/vscode-terosHDL #73: Feature : Instanciate module

suzizecat, 2021-02-04:
Hi,

While terosHDL seems amazing (I just started testing it), It seems to lack an important feature which would be an "Instanciate Module" command. This command would let you select a module and generate an instanciation snippet, presenting all ports so the user only have to "connect" the wires.

You could also do that using the name of the file where the module is instanciated as done in Sublime Text 3 (https://sv-doc.readthedocs.io/en/latest/instance/)

This feature is, IMHO, a must-have, and it would be great to see it in terosHDL !

Best regards,
Julien FAUCHER

suzizecat, 2021-02-04:
Hi!

It indeed is he feature I was loking for, thanks !
I didn't found it in the command palette, maybe you could consider adding this feature as a command ? (Or maybe I just missed it too)

qarlosalberto, 2021-02-04:
Yes! We can add it to the command palette.

qarlosalberto, 2021-02-05:
You can search in the command palette "generate template".

closed 2021-02-05 by qarlosalberto