TerosTechnology/vscode-terosHDL #76: Offline installation not works

Godhart, 2021-02-11:
Hello! I found that installation in isolated environment (not connected to internet) not works.
Plugin installed from XCI.
I've checked plugin installation folder - content almost matches to same folder on internet connected instance.

Plugin seems to be installed, but any action produces error. For example - for `Tools configuration` following error reported: `command 'teroshdl_tree_view.config' not found`

Would you please help to resolve it?

qarlosalberto, 2021-02-11:
Hi @Godhart 

It's really weird. The installation doesn't need a internet connection (or I suppose it). What version are you using and Where have you download it?

Godhart, 2021-02-11:
It's 0.1.3, downloaded from VS Marketplace (https://marketplace.visualstudio.com/items?itemName=teros-technology.teroshdl)
VSCode is 1.52.1 user setup if it does matters

Godhart, 2021-02-11:
Just tried to do same thing VM and result repeated.
After plugging VM to internet I've got following message:
```
Cannot activate the 'TerosHDL' extension because it depends on the 'Test Explorer UI' extension, which is not installed. Would you like to install the extension and reload the window?
```
It looks like not all dependencies are installed with plugin

Godhart, 2021-02-11:
Installing `Test Explorer UI` helped (at least configuration dialog works), will look further.

qarlosalberto, 2021-02-11:
Ahhh yest, that's right. This dependency will be removed in the next version. It's deprecated.

smgl9, 2021-03-29:
@Godhart The dependency has been removed in the current release. The issue should have been resolved but let us know if there are still issues with offline installation.

Thank you :)

closed 2021-03-29 by smgl9