TerosTechnology/vscode-terosHDL #77: 'export' is not recognized

freyn42, 2021-02-17:
When I try to run VUnit tests I get the following error output:
```
export  VUNIT_RIVIERAPRO_PATH=C:\Aldec\Riviera-PRO-2020.04\bin\riviera.exe;export  VUNIT_SIMULATOR=rivierapro;cd c:\Users\<username>\repos\.vscode\..;python run.py  "Not found." --no-color -x teroshdl_out.xml --exit-0   -v 'export' is not recognized as an internal or external command,
operable program or batch file.
```
It seems it's using `cmd ` in the background where `export` is not supported. Is there a way to change the console used or change the `export` to `set` or something?

My setup:
OS: Windows 10
vscode version 1.46.1
terosHDL version v0.1.3

qarlosalberto, 2021-02-17:
This problem is fixed in the new version. It will be release this week.

qarlosalberto, 2021-03-11:
We have released a new version. Can you check if it's fixed?

closed 2021-04-03 by qarlosalberto