TerosTechnology/vscode-terosHDL #81: Analysis doesn't seem to work with WSL Remote in VS Code

hughesjs, 2021-03-03:
I'm using WSL remote and my Makefile uses these flags:

`GHDLFLAGS=--std=08  --workdir=./$(TMPDIR) --warn-no-hide`

where `TMPDIR=build`.

I've then got TerosHDL set with these GHDL settings:

[image]

But I get unit "x" not found in library "work" almost everywhere despite the fact that running my makefile works just fine so there's no GHDL errors.

Any ideas?

hughesjs, 2021-04-11:
This might not actually be an issue with WSL as I'm now having the same issue on an arch installation.

The linter doesn't seem to be detecting work.x includes...

[image]

This is even after setting up a work library in the project manager:

[image]

My linter settings look fine to me:

[image]

Am I missing something?

qarlosalberto, 2021-04-11:
It works fine, GHDL only lints the current file.
You can try the lastest beta version with RustHDL:
https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
Disable the linter (none) or the errors will be duplicated.

hughesjs, 2021-04-11:
Ahh okay... that seems like something that would be useful to have

Does RustHDL lint the entire project?

qarlosalberto, 2021-04-11:
Yes!

hughesjs, 2021-04-11:
Dude, that's so much better! I love it! Great job!

Thanks so much!

closed 2021-04-11 by hughesjs