TerosTechnology/vscode-terosHDL #83: VHDL: some comments are not integrated to module documentation

lepommosaure, 2021-03-10:
Hi everyone,

First of all: Thank you for your dedicated work bringing the world this great extension!

I am running into the following issue when using "Module documentation preview" with VHDL files: some comments are ignored
I am using **Windows 10**

Typically, the following comments will not show up in the module documentation at all:
- The last signal declaration 
- The first process description in the architecture

I am joining an example source file and a screenshot of the output

[test_file.zip](https://github.com/TerosTechnology/vscode-terosHDL/files/6114536/test_file.zip)

[image]

qarlosalberto, 2021-03-11:
Thanks @lepommosaure I will fix it :)

davemaster, 2021-03-11:
> Thanks @lepommosaure <https://github.com/lepommosaure> we will fix it :)

thank You

holzi-x, 2022-12-22:
In addition to the fix I can state, that for example process descriptions are all included if the first line after the 'begin' keyword of an architecture is not a process definition but let's say a signal assignement. Obviously the parser needs one line not containing a known keyword to work correctly

qarlosalberto, 2023-06-26:
fixed on v5.0.0

closed 2023-06-26 by qarlosalberto