TerosTechnology/vscode-terosHDL #87: Extension host issues

ZirconfleX, 2021-03-24:
Hi,

On regularly base I get a popup about an extension host being unresponsive
[image]

The popup lets you restart the server or go to developer mode.
In developer mode the errors as showed in attached figure pop out.

What can be wrong?

Thanks

qarlosalberto, 2021-03-24:
Hi! Could you attach an image with the previous output to "Extension host terminated..." ?

ZirconfleX, 2021-03-24:
Dear Carlos,

This is the popup I get:

I copied all text for the errors in teh developers tools terminal
     Took screenshots because else it's unreadable.

Hope this si what you are looking for,

Regards,

Marc

qarlosalberto, 2021-06-05:
Check the beta version.

ZirconfleX, 2021-06-08:
I did, and it seems to work!
Guess this can be closed.

qarlosalberto, 2021-06-08:
Nice!

closed 2021-06-08 by qarlosalberto