TerosTechnology/vscode-terosHDL #89: Issues with VHDL documenter

GlenNicholls, 2021-03-24:
I am noticing the following problem with the VHDL documenter. I have this VHDL description for my entity:

```
--! This module takes single- or multi-cycle strobe and will generate a pulse on each rising
--! and falling edge. This will also provide signals that denote whether there
--! is a rising or falling edge of the strobe. This can be used in place of 1-shot signaling
--! and is also useful for passing acknowledgements between different clock domains.
--!
--! glitches on the output are possible when there are no registers, so it is recommended
--! to have at least 1 register to remove glitches from the output of the LUTs
```

Now, it produces the following documentation:
[image]

I have line breaks in the VHDL to align with a maximum line length I set in VS code. As you can see in the image, the documenter takes all of my line breaks literally and it hurts the readability in the documentation as not only does it not provide an extra blank line with just `--!`, but it breaks up the first paragraph with new lines. Can you add a setting for the tool such that when enabled, it will only insert line breaks in the documentation  when it sees an empty `--!` comment?

Now take the following example:

```
--! some documentation that goes on and on...
--! should not be a line break in the description.....
--!
--! Now there should be a line break.
```

This looks like this which is still wrong....
[image]

suzizecat, 2021-03-25:
I can confirm the same behavior with (System)Verilog. It looks like the markdown interpreter handle an empty line as a single line return and vice-versa.

GlenNicholls, 2021-03-25:
Yep, I also found the following:

```
--! This module takes single- or multi-cycle strobe and will generate a pulse on each rising
--! and falling edge. This will also provide signals that denote whether there
--! is a rising or falling edge of the strobe. This can be used in place of 1-shot signaling
--! and is also useful for passing acknowledgements between different clock domains.
--! `this is code` sample
--! | Tables | Are | Cool |
--! |--------|:--------:|--------:|
--! | col 1 is |  left-aligned | $1600 |
--! | col 2 is | centered | $12 |
--! | col 3 is | right-aligned  | $1|
--!
--! ```
--! library something;
--! use something.pkg.all;
--! ```
entity PulseGen is
    generic (
        NumRegs : natural := 0
    );
    port (
        --! clock
        clk   : in std_logic;
        reset : in std_logic; --! reset

        i_strobe  : in  std_logic;
        o_strobe  : out std_logic;
        o_pulse   : out std_logic;
        o_rising  : out std_logic;
        o_falling : out std_logic
    );
end entity PulseGen;
```

Here is the output
[image]

It appears that the code block gets sucked into the table even though the table has ended. Notice below how the markdown generation is also messed up (the code block gets sucked up to the same line as the final "|"):
# Entity: PulseGen
## Diagram
[image]
## Description
 This module takes single- or multi-cycle strobe and will generate a pulse on each rising
 and falling edge. This will also provide signals that denote whether there
 is a rising or falling edge of the strobe. This can be used in place of 1-shot signaling
 and is also useful for passing acknowledgements between different clock domains.
 `this is code` sample
 | Tables | Are | Cool |
 |--------|:--------:|--------:|
 | col 1 is |  left-aligned | $1600 |
 | col 2 is | centered | $12 |
 | col 3 is | right-aligned  | $1| ```
 library something;
 use something.pkg.all;
 ```

Instead, it should look like this
# Entity: PulseGen
## Diagram
![Diagram](PulseGen.svg "Diagram")
## Description
 This module takes single- or multi-cycle strobe and will generate a pulse on each rising
 and falling edge. This will also provide signals that denote whether there
 is a rising or falling edge of the strobe. This can be used in place of 1-shot signaling
 and is also useful for passing acknowledgements between different clock domains.
 `this is code` sample
 | Tables | Are | Cool |
 |--------|:--------:|--------:|
 | col 1 is |  left-aligned | $1600 |
 | col 2 is | centered | $12 |
 | col 3 is | right-aligned  | $1| 
 ```
 library something;
 use something.pkg.all;
 ```

GlenNicholls, 2021-03-25:
Instead, it should look like this
# Entity: PulseGen
## Diagram
[image]
## Description
 This module takes single- or multi-cycle strobe and will generate a pulse on each rising
 and falling edge. This will also provide signals that denote whether there
 is a rising or falling edge of the strobe. This can be used in place of 1-shot signaling
 and is also useful for passing acknowledgements between different clock domains.
 `this is code` sample
 | Tables | Are | Cool |
 |--------|:--------:|--------:|
 | col 1 is |  left-aligned | $1600 |
 | col 2 is | centered | $12 |
 | col 3 is | right-aligned  | $1| 
 ```
 library something;
 use something.pkg.all;
 ```

qarlosalberto, 2021-03-29:
@suzizecat @GlenNicholls could you check the line break issue in the lastest version?

https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

suzizecat, 2021-03-30:
@qarlosalberto on my side, the new line issue is fixed, however tables are no longer rendered with the following code :
```
//! `this is code` sample
//!
//!| Tables | Are | Cool |
//!|--------|:--------:|--------:|
//!| col 1 is |  left-aligned | $1600 |
//!| col 2 is | centered | $12 |
//!| col 3 is | right-aligned  | $1|
//!
//! ```
//! library something;
//! use something.pkg.all;
//! ```
//!
//! 
```

I tried to skip line between each row of the table, without success.
The space between `//!` and the table have no influence.

qarlosalberto, 2021-04-03:
Check the lastest release.

closed 2021-04-14 by qarlosalberto