TerosTechnology/vscode-terosHDL #90: Documentation issue with images

GlenNicholls, 2021-03-24:
In the documentation at https://terostechnology.github.io/terosHDLdoc/features/template_generator.html, I am trying to follow along with the examples to try things out while I'm testing terosHDL. However, it is really hard to do so since all of the snippets of code are only embedded in the image, but not in a code block near the image pointing things out. In these areas, can you please update the documentation to also include the snippet of code pictured to make it easier to copy and modify?

I think this is important because when I use tools like VUnit and other references, I often learn where in the documentation to find things, and instead of memorizing everything, I copy/paste and modify as needed. My suggestion will not only help those first learning the tool but also people who haven't used the tool enough to have all the most commonly used features memorized.

Reference the VUnit documentation as it provides the code snippets even when GIF's or images are used.

qarlosalberto, 2021-06-01:
I would nice to create a Github repo with examples, and to change images by code in the doc page.

GlenNicholls, 2023-08-11:
Sorry for the delay. 

If you reference https://terostechnology.github.io/terosHDLdoc/docs/editor/go_to_definition currently, the VHDL and Verilog code is embedded in the images/GIFs in the docs (ref below image):

[image]

My proposal here is to have those example code snippets available on the page such that a user can copy them and test those features exactly as they're presented in the documentation via your GIFs. 

For example, I am looking for VHDL/Verilog code blocks that can be copied like Python does in their documentation:

[image]

Nothing crazy here, I just want to be able to test out new language support features identically as they appear in the documentation/release notes in vscode. The easiest way to do that is to copy a code block example in your documentation that matches the feature you're presenting.

When code is embedded in images/GIFs in your release notes/documentation, I have to either 

A) write code that resembles your example to reproduce the new feature
B) I have to manually type out your example (line-by-line) which can be difficult when your screen movements like scrolling are fast. 

I trust that the new features work, but it would be much easier if your documentation included the code snippets such that the user could copy them to test the new features.

GlenNicholls, 2023-08-11:
To be clear, what I am asking for is that the code snippets used in the examples are copy-able on the page where they are presented.