TerosTechnology/vscode-terosHDL #91: Setting Python Path on Win 10

alperyazar, 2021-03-25:
Hi there,

Thanks for this fantastic plugin. I am experimenting with it on Win 10. I can't use *Dependencies Viewer* because it can't find Python. I have installed Python 3.9.2 and it's on the path, i.e. I can run `python` on cmd or PowerShell. I also tried to manually enter the path in extension settings but i didn't work. If I have to enter the path manually, how should I enter the python path, `C:\Python\python.exe` or `C:\\Python\\python.exe` (with escaping), etc.? 

Thank you.

alperyazar, 2021-03-25:
My mistake, `C:\\Python\\python.exe` works. I had typo in the settings and noticed after opening the issue. It still doesn't work even `python` is in the path and explicit set is required but double `\` does the trick.

The issue can be closed if it is OK for developers.

smgl9, 2021-03-25:
@alperyazar Thanks for the feedback and thanks for the workaround!
I left the issue open because the "trick" shouldn't be necessary. It happens in windows only.

alperyazar, 2021-03-25:
@smgl9 I agree, thank you!

monk31, 2021-06-08:
hi ,
i have the same problem with windows10, my version is 0.1.4
the rule is double quote + python.exe at the end of the string ?

alperyazar, 2021-07-20:
> 
> 
> hi ,
> i have the same problem with windows10, my version is 0.1.4
> the rule is double quote + python.exe at the end of the string ?

Hi,
Sorry for the late response. I don't have access to the machine right now but as far as I remember, yes, like ``C:\\Python\\python.exe``

Hqf2013, 2021-08-01:
I also encountered the same problem. The python setting is invalid under win10.

qarlosalberto, 2021-08-01:
We are working in the new version to fix the Windows issues.

ziyangd, 2021-12-20:
hi I have the same issue under win10, double / does not work.

qarlosalberto, 2021-12-20:
Have you triend double \ ?

ziyangd, 2021-12-21:
> 
yes I tried double \ and python is in my path. So where could be the problem?  
[image]
[image]

mlogelect, 2022-02-13:
Same problem here, i have python in my path and i have installed everything, but this is what i get
[image]

qarlosalberto, 2022-02-13:
@mlogelect open from cmd the python console. Type "import vunit", does it work?

mlogelect, 2022-02-13:
I tried and couldn't see the vunit. So i re-installed python3 and searched the path in command line again. This time it shows me a different path than before. Now everything works, thank you!

qarlosalberto, 2022-02-13:
Cool! I don't use Windows, so it's more difficult for me test it...