TerosTechnology/vscode-terosHDL #92: Errors reported who are not errors

ZirconfleX, 2021-03-25:
First of all, nice extension to vcode for VHDL development. 
I really like the documentation functions of the extension.

OS: Linux-Mint 20.4 Cinnamon
Vcode version: 1.54.3
TerosHDL version: 0.1.4
Using: VHDL

Settings for TerosHDL:
	 Python 3.9.2 and vunit-hdl installed
	Linter: xvhdl (path: /opt/Xilinx/Vivado/2020.2/bin)

VHDL components instantiated using "entity instantiation" method :

```
library MyComps_Lib
	use MyComps_Lib.all;

entity Top is (
	.......
);
end Top
--
architecture top_arch of Top is

constant .....
signal ......
attribute .....

begin
.......
Top_I_CmpOne : entity MyComps.CmpOne 
	generic map (
		.......
	)
	port map (
		......
	);
.......
end Top_arch;
```

The error reported is:

[VRFC 10-2987] 'cmpone' is not compiled in library 'mycomps_lib' 

Full view of the reported error.

{
	"resource": "/mnt/Libraries/SMyComps_Lib/Vhdl/CmpOne.vhd",
	"owner": "_generated_diagnostic_collection_name_#1",
	"code": "VRFC 10-2987",
	"severity": 8,
	"message": "[VRFC 10-2987] 'cmpone' is not compiled in library 'mycomps_lib'",
	"source": "TerosHDL:xvhdl",
	"startLineNumber": 284,
	"startColumn": 1,
	"endLineNumber": 284,
	"endColumn": 1.7976931348623157e+308
}

It seems as the linter using xvhdl only checks the current opened in an editor VHDL but does not go down in hierarchical structures of components used in the opened vhdl file.

Can this be solved by providing a file with the library structure of the design?
How does that file need to look and where does it need to be placed?

qarlosalberto, 2021-03-25:
Hi! Yes, for now the linters only checks the current file. We're working in the cross-file linter.

ZirconfleX, 2021-03-26:
Hi,

Ok, perfect.
If you're working on it, then I can close this.

closed 2021-03-26 by ZirconfleX