TerosTechnology/vscode-terosHDL #93: Documenter does not support Export PDF on Windows

GlenNicholls, 2021-03-25:
I tried to export the documentation as PDF and I get an error stating `Currently this feature is only supported in Linux`

qarlosalberto, 2021-04-03:
We use this library: https://www.npmjs.com/package/markdown-pdf
But it's deprecated. Do you know any portable program (C++, Rust..) to convert from Markdown/HTML to PDF?

GlenNicholls, 2021-04-15:
@qarlosalberto Hmmm, not that I can think of. For documenting our SW we use asciidoc, so it's not a big deal. I'm planning on using https://matthewsetter.com/technical-documentation/asciidoc/convert-markdown-to-asciidoc-with-kramdoc/ anyways to convert the markdown to asciidoc to be consistent with our SW flow and having everything get generated in CI.

qarlosalberto, 2021-07-01:
We have removed the pdf feature. It's very difficult to maintain and there is a lot of tools to cover to pdf.

closed 2021-07-01 by qarlosalberto