TerosTechnology/vscode-terosHDL #95: Preview markdown in documenter

GlenNicholls, 2021-03-25:
When writing the documentation, it would be nice to render a preview of the markdown version. Below are the current options when viewing the documentation:

[image]

I would like to have a preview markdown option such that I can see what the markdown looks like in real-time instead of just HTML.

GlenNicholls, 2021-03-25:
With the markdown extension (Markdown All In One extension), I have this option and it'd be nice to do something similar:
[image]

qarlosalberto, 2021-03-29:
I think that the HTML preview and the Markdown preview is the same. What is the difference? We convert the markdown to HTML, so it should be the same.

I think that Markdown Extensio does the same. Really you are seeing a HTML page with Markdown CSS. Or that I think...

GlenNicholls, 2021-03-29:
@qarlosalberto Sorry about the confusion, I noticed what I did above and I can see how that is misleading. When you look at the first image in the issue description, there is no way to preview the markdown version before exporting it within that same window. In my second screenshot, I clicked "Export markdown", opened the `.md` file, and used a different VS code extension to preview what the Markdown documentation looked like. The core aspect of this issue is I would like the ability to preview the markdown version in the same window that the HTML preview is in.

qarlosalberto, 2021-03-30:
Do you want to see the raw Markdown file or the rendered Markdown preview? 

Really you are seeing in the Documenter the rendered Markdown preview. Markdown preview should be the same to HTML preview. Perhaps there are differences such as the position of a table. But they are very similar.

GlenNicholls, 2021-03-30:
The rendered Markdown preview would be ideal. I just prefer that as Markdown is prettier IMO and for our documentation. We would be exporting the .md for our repos and not the HTML, so it seems natural that we would be able to view it as the Markdown preview since that's how we would save it.

qarlosalberto, 2021-05-27:
I think that there isn't an official markdown css style. Different md renders use different styles. We convert from raw markdown to html with a css style.

[image]

[image]

GlenNicholls, 2021-05-27:
That's a bummer. I use https://code.visualstudio.com/docs/languages/markdown, so would it be possible to open the .md file then? That way I can select to open the file and use that extension to preview the markdown.

I'm not set on this feature, though. If you don't think this is something that adds any value, I'm okay closing it.

qarlosalberto, 2021-06-01:
I think that the best option would be to an option to choose the CSS style.

qarlosalberto, 2021-07-01:
Done. Now we use GitHub markdown style: 
https://terostechnology.github.io/teroshdl-documenter-demo/

closed 2021-07-01 by qarlosalberto