TerosTechnology/vscode-terosHDL #97: Issues with template generator for file with multiple entity declarations

GlenNicholls, 2021-03-25:
I have a file with multiple entity declarations which is a style some people at my company used in legacy code. However, I noticed that when selecting "Copy as instance", it only creates the entity instantiation template for the first entity in the file. This is likely a similar story for the other template options. IMO, I think that the TerosHDL extension should refactor this.

I think a better feature than "Copy as instance" or "Copy as component" as it's currently implemented would be to provide a prompt when right-clicking the entity. When I say that, take the following code snippet:
```
entity PulseGen is
    generic (
        NumRegs : natural := 0
    );
    port (
        --! clock
        clk   : in std_logic;
        reset : in std_logic; --! reset

        i_strobe  : in  std_logic;
        o_strobe  : out std_logic;
        o_pulse   : out std_logic;
        o_rising  : out std_logic;
        o_falling : out std_logic
    );
end entity PulseGen;
```

If I were to right click `PulseGen`, it would be nice if the "Copy as instance" and "Copy as component" were options in the right-click menu shown below:
[image]

This way, I could map those key-bindings.

In short, I propose removing the following 2 entries in the command pallet:
[image]

and adding them as right-click options when the user right-clicks an entity name in the previous screenshot.

GlenNicholls, 2021-03-25:
EDIT: now that I think about it, I think all the template generator options should be selectable when right-clicking an entity.

GlenNicholls, 2021-03-30:
I am also suspicious that files with multiple entity declarations will not work with the documenter. I have not tested this, though.

qarlosalberto, 2021-06-01:
For now we only support one entity declaration. It's very tedious to change the parse, so it could be implemented in the future, but not now.