TerosTechnology/vscode-terosHDL #102: begin..end in style

shivampotdar, 2021-03-29:
Several Style guides (https://www.systemverilog.io/styleguide, https://www.systemverilog.io/styleguide) recommend `begin` being in the same line as the statement (always, if etc.) but it seems like all options with TerosHDL default to adding begin to a newline. 
If we could have a new style with this option or just another setting parameter it would be a great addition. 
Thanks!

suzizecat, 2021-03-29:
Would you mind specifying which formatter are you using ?
You can find it under "File > Preferences > Settings" then in the search bar you can type "Formatter Verilog Engine".

You are looking for the `Formatter > Verilog : Engine` field.

shivampotdar, 2021-03-29:
Thank you for the reply. I do not see an exact match to "Verilog Engine" in my settings.
The closest is `Formatter > Verilog > Type : A` which is set to `iStyle`

suzizecat, 2021-03-29:
You might want to try to set the indentation style to "Kerningan&Ritchie" which should follow those recommendations.

shivampotdar, 2021-03-29:
Thank you for the suggestion. Even with that option I see the same behavior.

qarlosalberto, 2021-03-29:
@suzizecat has added a new formatter. It will released in the next version

suzizecat, 2021-03-29:
@qarlosalberto However, it won't really solve this issue (even though it simply doesn't change the position of the begin/end tags)
iStyle with K&R style _should_ have done the trick. I will have a look at that when I got time.

qarlosalberto, 2021-04-03:
There is a bug. It will be fixed in the next release.

shivampotdar, 2021-04-05:
That's good to know. Thanks @qarlosalberto!

qarlosalberto, 2021-04-14:
Fixed in the lastest beta version.

closed 2021-04-14 by qarlosalberto