TerosTechnology/vscode-terosHDL #104: Format result not applied to current document.

suzizecat, 2021-03-29:
Under unknown conditions, formatter output seems to not be pulled back from temporary file to current editor.
Seen using S3SV formatter, under RH7 on a compute farm. 
No particular message shown in the debug tools. 

Some checks should be added to notify the user if something went wrong in the process.

qarlosalberto, 2021-03-29:
Also, It would be necessary a python3 path checker. It will be common to other features, as dependencies viewer.

qarlosalberto, 2021-04-15:
@suzizecat is it fixed?

suzizecat, 2021-04-16:
I have this issue on windows. Investigating what's going on.
As far as I can currently see, the temp file is created with file content but the formatter isn't ran on it.

What's weird is that I have a message in the debug tool which print the command used. If I copy-paste this command in a terminal, the temp file is formatted as intended.

PR #133 (2021-04-23): Use async/await in get_options python resolution
Fix #104
commit c157e122 (2021-04-23): Use async/await in get_options python resolution (#133)

Fix #104
closed 2021-04-23 by qarlosalberto