TerosTechnology/vscode-terosHDL #109: Dependency graph is not fully recomputed upon adding a new file.

suzizecat, 2021-04-02:
Given the following expected dependency graph : 

[image]

In the dependency viewer :
 - Adding all files in a batch works as expected.
 - Adding C and C1 **before** A, B and B1 as a batch works,
 - Adding C and C1 **after** A will not link them to A.

This behavior showed up using only VHDL files.
It was not tested with SV files.

qarlosalberto, 2021-04-03:
Could you share an example? I can't reproduce it with my files.

qarlosalberto, 2021-04-15:
> Could you share an example? I can't reproduce it with my files.

@suzizecat

qarlosalberto, 2021-04-15:
An example with the files.

suzizecat, 2021-04-29:
It seems that I cannot reproduce anymore (at least I haven't saw the issue in quite some time) so you may close the issue and if I find  a way to reproduce accurately I'll reopen this issue or open a new one.

closed 2021-06-01 by qarlosalberto