TerosTechnology/vscode-terosHDL #117: [bug] Issues with RustHDL .vhdl_ls.toml on Windows

GlenNicholls, 2021-04-07:
I am seeing the following problems with the `.vhdl_ls.toml` that is produced in the user home directory on Windows:

```
'\C:/Xilinx/Vivado/2019.2/data/vhdl/src/unisims/unisim_VCOMP.vhd',
```

Teros is prepending a `\` in front of the Windows drive letter (`\C:`) which is not correct. Instead, all files in the list should look like this:

```
'C:/Xilinx/Vivado/2019.2/data/vhdl/src/unisims/unisim_VCOMP.vhd',
```

Another issue is there is a mixture in case for the drive letter. E.g. sometimes it uses a lower-case `c` instead of the proper notation of `C`. Functionally this shouldn't make a difference, but Teros should be consistent about it, ideally upper-case.

The next problem I am seeing is that teros is duplicating files in the `.vhdl_ls.toml` as well. This occurs when I add the same CSV that is attached via `Load files from file list` to the same project multiple times. What this does is append the same files to all of the libraries in the `.vhdl_ls.toml` instead of ensuring that it is not a duplicate. RustHDL doen't like duplicate files and instead of reporting that it is already there when it happens, you have to walk through the files one-by-one until you see the error that it is a duplicate. My proposal for this problem is that TerosHDL over-write the `.vhdl_ls.toml` whenever the file list changes. Either if it detects a change in the file list CSV file or if a new file list is loaded.

For all of the above issues, you can see them in the attached files. the CSV is my input file and the vhdl_ls file is the `.vhdl_ls.toml` file Teros saves in my home directory.

These are the below bugs:
* [ ] on windows, fix `\` before drive letter
* [ ] overwrite the `.vhdl_ls.toml` whenever the CSV file changes or whenever a new file list is loaded

[files.zip](https://github.com/TerosTechnology/vscode-terosHDL/files/6274920/files.zip)

qarlosalberto, 2021-04-08:
Thank you for the report! :)

I think that the best solution is:
1. Don't add the file if it's already added. (maybe you have created the project manually and you add from the CSV only the common libraries). So we prevent the duplicate files.
2. Refresh the project always you add a CSV.

What do you think?

GlenNicholls, 2021-04-08:
> Thank you for the report! :)
>
> I think that the best solution is:
>
> 1. Don't add the file if it's already added. (maybe you have created the project manually and you add from the CSV only the common [...]

1. I think re-writing the TOML whenever you select a CSV in a project is a better solution. At least for me, I generate the CSV file programmatically so it sounds like more work on your end to keep track of what is currently there and basically diff-adding files.
2. Yes, I agree

Regarding 1., I personally don't have a preference between your suggestion and mine. When I generate the CSV, it gets all common libraries anyways so I don't have a need for you to diff-add or diff-remove files in a library, it just sounds like more work.

closed 2021-04-14 by smgl9