TerosTechnology/vscode-terosHDL #118: [bug] Issues with file explorer opening location on Windows

GlenNicholls, 2021-04-07:
TerosHDL, whenever it opens the file explorer on Windows to browse files, it opens it in the directory of the current file selected in the editor. 

E.g. if I open the file `C:\tmp\something\file.py` in VS code to edit and then I navigate to the project explorer and select `Add file` for the current project followed by `Load files from file list`, teros opens the file explorer in `C:\tmp\something\`. This is not very ergonomic as when I'm using the project manager, everything I want is in a repository, usually on a different drive than some python script I was working on that is unrelated to the project. 

I propose that whenever TerosHDL opens the file explorer, it should be in the workspace root.

qarlosalberto, 2021-04-08:
I'm not sure what is the best option, it isn't a bug.
What happen if you have 4 folders? What workspace folder should TerosHDL choose? It could be an option:

GlenNicholls, 2021-04-08:
>What happen if you have 4 folders? What workspace folder should TerosHDL choose? It could be an option:

I'm actually not sure, to be honest, I see what you're saying though. In the case that the selected file is in a workspace, I think it should be opened in that workspace root. However, if you have 4 workspaces and the selected file is not in any of them, I'm not sure where it should open. Maybe in the home directory?

I just tested opening a file in Xilinx Vivado, it always defaults to the home directory for the first time. After the first time you open a file, it defaults to the previous location.