TerosTechnology/vscode-terosHDL #121: IEEE Libraries Not Detected

hughesjs, 2021-04-11:
I've got this project setup:

[image]

And running it via the start simulation button throws this error:

```
export  VUNIT_GHDL_PATH=;export  VUNIT_SIMULATOR=ghdl;cd /home/james/repos/ee3dsa-digital-systems-architecture/src/interpreter;python3 dsalang_interpreter.vhd   --no-color -x teroshdl_out.xml --exit-0      File "/home/james/repos/ee3dsa-digital-systems-architecture/src/interpreter/dsalang_interpreter.vhd", line 1
    library ieee;
            ^
SyntaxError: invalid syntax
```

I can perform the build with my makefile just fine so I'm not sure what's going wrong here...

qarlosalberto, 2021-04-11:
It seems an error in the GHDL path configuration. Could you set the GHDL
path in the configuration?

hughesjs, 2021-04-11:
I've set it:

[image]

[image]

But still seem to be getting the same error

qarlosalberto, 2021-04-11:
Could you share the output? Exactly the same?

hughesjs, 2021-04-11:
I'd just edited it before you replied :)

qarlosalberto, 2021-04-11:
Ahh I see. You only need to add in the project the VUnit "run.py" and
select it as top.

hughesjs, 2021-04-11:
Ahh okay... so does that start simulation button run the run.py testbenches rather than the simulation of the top entity then?

qarlosalberto, 2021-04-11:
Yes. For now only VUnit is supported. In the future we will add baremetal
simulations.

qarlosalberto, 2021-04-11:
https://terostechnology.github.io/terosHDLdoc/features/project_manager.html

hughesjs, 2021-04-11:
Ahh okay, might I suggest adding a warning if the user tries to run a top file that isn't a run.py? 

Thanks for your help tonight though!

closed 2021-04-11 by hughesjs

qarlosalberto, 2021-04-11:
Yes, it's a good idea :)