TerosTechnology/vscode-terosHDL #135: sometimes，the statemachine viewer could not find the statemachine

qgzln, 2021-04-26:
[image]
As shown in FIG，There is no state diagram, although the text description says that a state machine was foundthere is no statemachine

qarlosalberto, 2021-04-26:
Could you share your file?

cbiker, 2021-04-29:
if always block without begin end (only if reset else case...) then it not found by viewer. 
` // edgeDetector.v
// Moore and Mealy Implementation

module edgeDetector
(
    input wire clk, reset, 
    input wire level, 
    output reg Mealy_tick, Moore_tick
);

localparam  // 2 states are required for Mealy
    zeroMealy = 1'b0,
    oneMealy =  1'b1;
    
localparam [1:0] // 3 states are required for Moore
    zeroMoore = 2'b00,
    edgeMoore = 2'b01, 
    oneMoore = 2'b10;

reg stateMealy_reg, stateMealy_next; 
reg[1:0] stateMoore_reg, stateMoore_next;

always @(posedge clk, posedge reset)
begin
    if(reset) // go to state zero if rese
        begin
        stateMealy_reg <= zeroMealy;
        stateMoore_reg <= zeroMoore;
        end
    else // otherwise update the states
        begin
        stateMealy_reg <= stateMealy_next;
        stateMoore_reg <= stateMoore_next;
        end
end

// Mealy Design 
//! fsm_extract
always @(stateMealy_reg, level)

    case(stateMealy_reg)
        zeroMealy: // set 'tick = 1' if state = zero and level = '1'
            if(level)  
                begin // if level is 1, then go to state one,
                    stateMealy_next = oneMealy; // otherwise remain in same state.
                    Mealy_tick = 1'b1;
                end
        oneMealy: 
            if(~level) // if level is 0, then go to zero state,
                stateMealy_next = zeroMealy; // otherwise remain in one state.
    endcase

// Moore Design
//! fsm_extract 
always @(stateMoore_reg, level)
begin
    // store current state as next
    stateMoore_next = stateMoore_reg; // required: when no case statement is satisfied
    
    Moore_tick = 1'b0; // set tick to zero (so that 'tick = 1' is available for 1 cycle only)
    case(stateMoore_reg)
        zeroMoore: // if state is zero,
            if(level) // and level is 1
                stateMoore_next = edgeMoore; // then go to state edge.
        edgeMoore:
            begin
                Moore_tick = 1'b1; // set the tick to 1.
                if(level) // if level is 1, 
                    stateMoore_next = oneMoore; // go to state one,
                else    
                    stateMoore_next = zeroMoore; // else go to state zero.
            end
        oneMoore: 
            if(~level) // if level is 0,
                stateMoore_next = zeroMoore; // then go to state zero.      
    endcase
end
endmodule `

qgzln, 2021-04-30:
> if always block without begin end (only if reset else case...) then it not found by viewer.
>
> ```
>  // edgeDetector.v
> // Moore and Mealy Implementation
>
> module edgeDetector
> (
>     input wire clk, [...]

Thanks,Follow your suggestion,I can find the statemachine in FIG

qarlosalberto, 2021-06-01:
@qgzln could you share you state machine that fails?

qgzln, 2021-06-02:
I found that this problem is caused by code irregularities

closed 2021-06-02 by qarlosalberto

cbiker, 2021-11-22:
>>I found that this problem is caused by code irregularities
what is mean? code of example or code of teros? 

now i see that not only always problem but its global problem with "begin end"
fsm recognized only if begin end at module at always and extra end at case(and code isnt right).

module pattern_generator_stm ( bla bla );
begin // if module not used begin end fsm is not recognized
enum logic[3:0] {bla} state;
//! fsm_extract
always @(posedge clk )
begin // if always not used begin end fsm is not recognized
case (state)
half_wave_2: if (controls_reg.pwm_factor == 0) //begin //if not used leaving state in fsm
if (wave_cnt == controls_reg.frequency_div-1)
if (perod_num_cnt == controls_reg.period_num) state = count;
else state = dead_time_2;
else state = dead_time_2;
//end //if not used leaving state in fsm
else
if (wave_cnt == controls_reg.frequency_div-controls_reg.pwm_factor-2) state = ddr_edge_2;
else state = half_wave_2;
//
ddr_edge_2: state = dead_time_2;
dead_time_2: if (wave_cnt == controls_reg.frequency_div-1)
if (perod_num_cnt == controls_reg.period_num) state = half_wave_2;
else state = half_wave_2;
else state = dead_time_2;
count: state = half_wave_2;
default: state = half_wave_2;
endcase
end // excess end is verilog error, but witout fsm is not recognized
end // if always not used begin end fsm is not recognized
end // if module not used begin end fsm is not recognized

endmodule