TerosTechnology/vscode-terosHDL #136: Modelsim Linter - vcom

skaester, 2021-04-28:
For Modelsim as Linter, the wrong path settings are used:
[image]

here the Result of the debugger:

[image]

The "undefined" at the end of the path (Debugger output) must be replaced by "vcom".

Nemmerle, 2021-05-11:
I have the same problem in a Windows 10 machine.

However, the problem is not shown in an OpenSuse Linux.

```bash
[colibri][info] Linting with command: vcom -quiet -nologo -2008 
```

qarlosalberto, 2021-05-11:
We will fix it in the next version.

closed 2021-10-05 by qarlosalberto