TerosTechnology/vscode-terosHDL #138: Can't find any state machine in SystemVerilog code using //! fsm_extract above always_ff and always_comb

henriquedemoura, 2021-04-30:
When I use 
```
//! Next state logic
//! fsm_extract
always_comb begin : next_state_logic
(...)
end
```
or
```
//! FSMD register block
//! fsm_extract
always_ff @(posedge clk, posedge reset) begin : fsmd_control
(...)
end
```
The state machine viewer can't find the machine inside of the always blocks.

qarlosalberto, 2021-04-30:
Could you share a complete code example?

henriquedemoura, 2021-05-01:
Here the code!

```
module module_test
(
    input logic clk,            //! core clock
    input logic reset,          //! assynchronous reset

    input logic enable,         //! my module enable signal!

    input logic [7:0] a1,       //! my first input
    input logic [7:0] a2,       //! my second input

    output logic [7:0] b1,      //! my first output
    output logic [7:0] b2       //! my second output
);

//! I am used to do this while I code state machines
typedef enum {zero, one, two, three} state_type;

state_type state_reg, state_next;

//! voilà something
logic temp;
assign temp = b1;

//! This is where I change states, let's put "fsm_extract" here only to see what happens
//! fsm_extract
always_ff @( posedge clk, posedge reset ) begin : registers
    if (reset) begin
        state_reg <= 0;
    end
    else begin
        state_reg <= state_next;
    end
end

//! This is my FSM
//! fsm_extract
always_comb begin : my_fsm
    state_next = state_reg;
    b1 = 1'b0;
    b2 = 2'b0;
    case(state_reg)
        zero: begin
            if (enable)  begin
                //! what happens if I comment here?
                state_next = one;
            end
        end
        one: begin
            b1 = a1 + a2;
            state_next = two;
        end
        two: begin
            b2 = temp + a1 + 2;
            state_next = zero;
        end
    endcase
end

endmodule
```

qarlosalberto, 2021-06-06:
Fixed in the beta version.

closed 2021-06-06 by qarlosalberto