TerosTechnology/vscode-terosHDL #139: Problem visualizing virtual buses on module documentation with SystemVerilog

henriquedemoura, 2021-05-03:
Maybe I'm doing this the wrong way, but I'd like your opinion on generation virtual buses in the documentation.

I can have virtual buses if I write the comments above each port line, and if I have other ports after the virtual bus with the same comment style.

I have this code example:

```
module test
  (
    input logic a1,         //! first input

    //! @virtualbus virtual_block1 @dir in
    //!
    //! second input
    input logic a2,
    //! third input
    input logic a3,
    //! @end

    //! @virtualbus virtual_block2 @dir out
    //!
    //! Second output
    output logic b2,
    //! Third output
    output logic b3,
    //! @end

    //! if you remove this port, the virtual buses will disapear
    output logic b4

  );

  assign b1 = a1;
  assign b2 = a2;
  assign b3 = a3;
  assign b4 = a1 & a2;

endmodule
```

For example, if the  `output logic b4` is removed but the comma in `output logic b3,` is still there, we have the virtual bus, but once this comma is removed (leaving us  `output logic b3`), the virtual bus "virtual_block2" disappears (on the documentation viewer).

I usually like to document my lines with side comments, but using them, I can not generate the virtual buses on the documentation viewer... Should I adopt the style of the first code example because it was meant to be like that, or this way should also work?

```
module test
  (
    input logic a1,         //! first input

    //! @virtualbus virtual_block1 @dir in
    //!
    input logic a2,     //! second input
    input logic a3,     //! third input
    //! @end

    //! @virtualbus virtual_block2 @dir out
    //!
    output logic b2,    //! Second output
    output logic b3,    //! Third output
    //! @end

    //! if you remove this port, the virtual buses will disapear
    output logic b4

  );

  assign b1 = a1;
  assign b2 = a2;
  assign b3 = a3;
  assign b4 = a1 & a2;

endmodule
```

smgl9, 2021-05-03:
https://github.com/TerosTechnology/colibri/issues/199

smgl9, 2021-05-03:
Thank you for your feedback :)
We will investigate how to add support for inline comments in virtual buses. And the bug in last line with `@end` is something to be solved for the next release.

closed 2021-05-27 by qarlosalberto