TerosTechnology/vscode-terosHDL #143: Nestlist always showed in loading

heartacker, 2021-05-17:
版本: 1.57.0-insider (system setup)
提交: 40d5e6796fbc32d67dd0009e5b0027003803fd7e
日期: 2021-05-14T20:13:58.234Z
Electron: 12.0.7
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Windows_NT x64 10.0.14393

```verilog
module twomux (out, a, b, sl);
  input a, b, sl;
  output out;
  not u1 (nsl, sl );
  and #1 u2 (sela, a, nsl);
  and #1 u3 (selb, b, sl);
  or    #2 u4 (out, sela, selb);
endmodule
```
[image]

qarlosalberto, 2021-05-20:
I will check it.

qarlosalberto, 2021-05-20:
Could you share the complete code? nsl isn't declared?

heartacker, 2021-05-22:
```verilog
//! hhhh
module MUX2_1 (out, a, b,                 sel);
  // Port declarations
  output out;
  input a, b, sel;
  wire out, a, b, sel;
  wire sel_, a1, b1;
  // The netlist
  not (sel_, sel);
  and (a1, a, sel_);
  and (b1, b, sel);
  or (out, a1, b1);
endmodule
```

FYI this code still has the problem,

dzx-dzx, 2021-05-25:
The problem also exists for all my projects.

dave-grabowski, 2021-05-29:
Same thing. I even tried to use the example `addbit`:
```verilog
module addbit (
    a      , // first input
    b      , // Second input
    ci     , // Carry input
    sum    , // sum output
    co       // carry output
    );
    //Input declaration
    input a;
    input b;
    input ci;
    //Ouput declaration
    output sum;
    output co;
    //Port Data types
    wire  a;
    wire  b;
    wire  ci;
    wire  sum;
    wire  co;
    //Code starts here
    assign {co,sum} = a + b + ci;
    
    endmodule // End of Module addbit
``` 

Is there any other dependency needed?

qarlosalberto, 2021-06-01:
We have a compatibility issue with the latest VSCode version.

smgl9, 2021-06-01:
https://github.com/TerosTechnology/vscode-terosHDL/issues/141

qarlosalberto, 2021-08-11:
Fixed in the beta version

closed 2021-08-11 by qarlosalberto