TerosTechnology/vscode-terosHDL #146: SystemVerilog Header files Support

degas91, 2021-05-20:
The tool does not support .svh (systemverilog headers) files. Please add syntax highlighting for .svh files.

smgl9, 2021-05-20:
Thank you for the feedback!

We will add support for shv files in the next release :)

degas91, 2021-05-21:
Thanks, I really love what you guys are doing! When can I expect the next release? 

Thanks again!

smgl9, 2021-05-21:
Next release will have many changes and we need free time to finish and test all of them until they are stable enough. So I can not guarantee a date. Sorry
I just can say that the new version is under development right now :)

Anyway I'll let you know when there is a beta version available with this feature in case you want to test it.

degas91, 2021-05-21:
Thanks , I would definitely like to test the beta version.

closed 2021-05-22 by qarlosalberto

smgl9, 2021-05-22:
@degas91 

The support for svh files should be available now in the beta release. you can install it manually: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

Anyway take into account that some functionalities could be broken due to the ongoing development :)