TerosTechnology/vscode-terosHDL #150: Remove a single file from project

smgl9, 2021-05-24:
In master branch, trying to remove a single file in the project manager removes all files in the active project.

suzizecat, 2021-07-22:
Are you sure ?
For me it seems to work correctly on master. However the removed file is not the one near which you clicked the "remove" button, but the one actually selected. So, if you select a library and hit "-" near a file, you'll actually remove the library.

qarlosalberto, 2021-07-22:
You are right. There are 2 solutions:
- Remove "-"
- If you click "-" remove only the file, not all selected files/libraries.

suzizecat, 2021-07-22:
If you can, in my opinion, the best way would be: 
 - If you have selected the item you are removing, then no problem, remove every selected items
 - Otherwise, only remove the item you have clicked "-" on.

qarlosalberto, 2021-07-22:
It's a very good idea! I will do it

qarlosalberto, 2021-08-11:
Fixed.

closed 2021-08-11 by qarlosalberto