TerosTechnology/vscode-terosHDL #154: an error will be reported when I create a new file

qgzln, 2021-05-26:
When I use vscode to open a Verilog file and instantiate it, then double-click to create a new file like
[image]
with the mouse and an error will be reported

[image]

qarlosalberto, 2021-06-01:
Could you check if you have the same problem in the beta version? 

https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/V0.0.5

qgzln, 2021-06-02:
> Could you check if you have the same problem in the beta version?
> 
> https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/V0.0.5

I have the same problem in the beta version

qarlosalberto, 2021-06-02:
Could you explain the problem in more detail?

smgl9, 2021-06-02:
> > Could you check if you have the same problem in the beta version?
> > https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/V0.0.5
> 
> I have the same problem in the beta version

This is the right link for the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
Just in case you downloaded an older version...

qgzln, 2021-06-02:
> Could you explain the problem in more detail?

when i use generate template to generate some template such as cocotb testbench etc,the template is on clipboard， so i want to creat new file , if i creat new file without any extension like .v(because i often creat new file through double-click with the mouse) , an error will be reported.
 1 
[image]
2
[image]
3 A few seconds later

[image]

qgzln, 2021-06-02:
> This is the right link for the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
> Just in case you downloaded an older version...

thank you ,I will try it right now

qgzln, 2021-06-02:
> This is the right link for the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
> Just in case you downloaded an older version...

the new have the  same problem

qgzln, 2021-06-02:
I find another problem using the new version ,
[image]
Nothing happens when you click the mark，but you can use it through ctrl+shift+p ，Call up the format command

qarlosalberto, 2023-06-26:
fixed on v5.0.0

closed 2023-06-26 by qarlosalberto