TerosTechnology/vscode-terosHDL #155: Dependency viewer fails to open

lukipedio, 2021-05-26:
I am getting this error:

[image]

while trying to open dependency viewer in TerosHDL 0.1.4.

My system path points to Python 3.9 installation.

I have tried also to add that path in the extension configuration, but no luck.

lukipedio, 2021-05-26:
ok. I'll check it on next release!

lukipedio, 2021-05-26:
I have installed the latest release beta (it appears to be v0.1.91).
It doesn't work yet. 
This is the message I get in the devtools: https://prnt.sc/13fulz4

qarlosalberto, 2021-05-26:
It's related to: https://github.com/TerosTechnology/vscode-terosHDL/issues/156

Go to TerosHDL settings and set your python3 path.

smgl9, 2021-05-26:
@lukipedio  In a windows shell type: `where python3` and copy the entire path in teroshdl python path settings, as @qarlosalberto  mentioned.
The path must be something like this: `C:\Users\isma\AppData\Local\Microsoft\WindowsApps\python3.exe`

lukipedio, 2021-05-26:
Unfortunately, it doesn't work in my case.
I still get an error about SVG file.
[image]
But if I save it in Markdown format or HTML, the figure is there in a good SVG output file.

qarlosalberto, 2021-05-26:
@lukipedio where do you have the error, in the dependency viewer?

lukipedio, 2021-05-27:
@qarlosalberto Yes, in the dependency viewer.

qarlosalberto, 2021-05-27:
Have you installed VUnit?

lukipedio, 2021-05-27:
My fault, sorry.
Too many Python installations.... 
I have installed vunit-hdl for the one which is returned by 'where python3' and now everything works just fine.

[image]

closed 2021-05-27 by lukipedio

qarlosalberto, 2021-05-27:
Good!